nordicsemi/IOS-nRF-Mesh-Library #430: Issues observed in configuring and controlling devices after importing a database.

dhaval-sabhaya, 2022-05-23:
**Describe the bug**
Hello Team,
I am stuck due to below describe issue in one of my Project 
Please help me to resolve this issue using nRF mesh SDK.
Your quick help will be appreciated.

**Issue:-** When I export my database from the same iPhone or 1 iPhone to another and try to operate the same get intensity operation for the individual controller from the iPhone then it does not work. I get a timeout error or error of getting bearer close.

**Note:-** I have updated the local provisioner's unicast address and tried to operate it but failed. also applied suggested solutions https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/issues/391
and follow a similar issue https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/issues/393
**But nothing works**

**To Reproduce**
Pre-Condition:
The section which contains a group should be created.

Case 1:
1) Configure light controllers in the group.
2) Export the database.
3) Now add the sensor device.
4) Import the database exported in step-2

Try to get and set the intensity of the light controllers.

Result:
In case 1, get operations results in request timeout, and setting intensity does not work. Even after relaunching the application, the issue is not resolved.

**Expected behavior**
Get intensity individual LC (node) should work as same before importing database

Please let me know in case of further details would require.

dhaval-sabhaya, 2022-05-24:
Hello @philips77
Thanks for your quick reply
Here are inline your asked questions

What "Section" are you talking about?

> We have created an intermediate section in the collection of groups, and assigned the app key and net key in the same flow.
Code Snippet:
public class DSection: Codable {
    internal weak var meshNetwork: MeshNetwork?
    
    public var idSection: Int
    public var nameSection: String
    public var networkKey: NetworkKey
    public var applicationKey: ApplicationKey
----

Are you talking about nRF Mesh app, or your own, which is using this library?
> We are using the library in our project

What "other device" in Case 1, point 3?
> other device means second, third, etc. after adding the first device

What does it mean to "reset mesh before importing" (you mean to forget the current configuration/create a new config?)
> create new config suggested nRF Mesh SDK sample code
        (UIApplication.shared.delegate as! AppDelegate).createNewMeshNetwork()
        
        Please let us know in case of further details would require.

smartfarming-udf, 2022-10-31:
@philips77 Can you please review our answers to your questions and advise if further info is required?

philips77, 2023-06-01:
Hello!
Sorry for the very long delay, I switched to another projects.
I hope you did find the answer you were looking for. I'll now close this issue as stale.

closed 2023-06-01 by philips77