nordicsemi/IOS-nRF-Mesh-Library #431: How many bytes can I send to a group at one time?

okappi0258, 2022-06-01:
I have created a group with approximately 20 modules to send messages. However, there is a delay among the 20 modules because the packets seem to be discarded. I think it is due to the large size of the message, how many bytes can be sent in one message? Right now I am sending 13 bytes.

philips77, 2022-06-03:
Have a look at Table 3.42 from Mesh Profile 1.0.1:

[image]
[image]

If a segmented message is sent to a group address each segment should be sent multiple times to increase the changes of successful delivery. The iOS mesh library does it for you:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/0f25a7127d1581d1f81e026742257dc08fa251f7/nRFMeshProvision/Classes/Layers/Lower%20Transport%20Layer/LowerTransportLayer.swift#L632-L660

Do you have the 20 nodes in proximity to each other, or the messages need to propagate through all to reach the last one?

okappi0258, 2022-06-03:
Thanks for the reply.
I would like to create a group with 20 nodes and send a message to the group address so that all connected LEDs light up at the same time. 20 nodes are in close proximity and the message will reach all nodes. However, there is a delay in the situation.

philips77, 2022-06-03:
If you'd use *Generic OnOff Set Unacknowledged* message, it has an optional parameter `delay`:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/0f25a7127d1581d1f81e026742257dc08fa251f7/nRFMeshProvision/Classes/Mesh%20Messages/Generic/GenericOnOffSetUnacknowledged.swift#L61-L72
The `delay` parameter can be used exactly for this purpose. You may send two messages, one following the other, where the first one has a `delay` set to the actual delay you use before sending the second one. Then, no matter which message is received by the remote nodes, they should turn on/off approximately together.

Example:
1. Send `GenericOnOffSetUnacknowedged(state, some transition, delay = 200)` // 200 * 5 ms = 1 sec
2. Wait 1 sec
3. Send `GenericOnOffSetUnacknowedged(state, some transition, no delay)`

philips77, 2022-06-03:
Also, there will always be a delay when you're using method like that. Some messages may need to be relayed, repeated, etc. But in practice this is not noticeable in small areas.

okappi0258, 2022-06-04:
The delay value is set to 0 for transmission. However, the delay still occurs.
In the message I am sending, the payload size has exceeded 11. Is that the cause?
If so, can I change the maximum payload size?

philips77, 2023-06-01:
Hi, sorry for 1 year delay. I must have missed your reply. I'll close the issue now, hoping that you have found the answer.