nordicsemi/IOS-nRF-Mesh-Library #433: In some cases messages are not sent correctly

hinzclaudio, 2022-06-08:
I have been working with this library for about 1.5 years and everything seemed to work fine. 

After some refactoring, I decided to move the MeshNetworkManager's delegate queue from the main queue to a background queue with a qos of .userInitiated. I have been experiencing issues ever since. What's especially weird about this, is that my issues arise randomly - mostly when sending to more than one node at a time.

### The Problem
When sending a message, in some cases I will never receive a response from the device. However, my MeshNetworkDelegate's functions are called correctly.

For example: 
- I try to send a GenericLevelSet message to a node. In this case I can see that the actual device has not received this message.
- However, the delegates `meshNetworkManager(_: didSendMessage: from: to:)` is called three times after sending. (Considering I have set the number of retries to two, this is correct.)
- Then, the delegates `meshNetworkManager(_: failedToSendMessage: from: to: error:)` is called.

The issue at hand is, the app actually never sends anything to the network. I can solve the issue only by restarting my app. Unfortunately I have not yet found a consistent strategy to reproduce the issue. Have you got any ideas where I should start debugging?

The bug is reproducible with about any device. We are mostly testing on iOS 15.

closed 2022-06-09 by hinzclaudio

philips77, 2022-06-10:
Hi, did you manage to solve the issue? Was it something on your end, or did you create a different issue?

hinzclaudio, 2022-06-10:
Hi, I actually found the bug on our end. Thanks for the quick reply!