nordicsemi/IOS-nRF-Mesh-Library #434: Sensor Cadence Set command always failing

bhautikVolansys, 2022-06-08:
Hello Team, 

I am trying to set cadence on sensor for supported property (Precise Present Ambient Temperature, Present Ambient Relative Humidity, Photosynthetically Active Radiation). But always I am getting error Request timed out. Can you help me to recognise what's wrong in payload?

Precise Present Ambient Temperature

Send  Message
message: sensorCadenceSet 
Sending SensorCadenceSet(property: Precise Present Ambient Temperature, cadence: SensorCadence(fastCadencePeriodDivisor: 1, statusTriggerType: 0, statusTriggerDelta: SensorCadence.StatusTriggerDelta.values(down: 0.10 °C, up: 0.10 °C), statusMinIntervalValue: 11, fastCadenceLow: -40.00 °C, fastCadenceHigh: 125.00 °C)) from: Primary Element (0x0001), to: 0100

Receive Message
meshNetworkManager failedToSendMessage > message: SensorCadenceSet(property: Precise Present Ambient Temperature, cadence: SensorCadence(fastCadencePeriodDivisor: 1, statusTriggerType: 0, statusTriggerDelta:  SensorCadence.StatusTriggerDelta.values(down: 0.10 °C, up: 0.10 °C), statusMinIntervalValue: 11, fastCadenceLow: -40.00 °C, fastCadenceHigh: 125.00 °C)) opCode 85 error Request timed out.

bhautikVolansys, 2022-06-08:
@philips77 Can you please take a look into this?

bhautikVolansys, 2022-06-10:
Hello Team, 

We found an issue in the SDK SensorCadence structure used for data binding for sensor properties and preparing payload encoded Data format conversion for "statusMinInterval" data appending.

SDK> Mesh Messages > SensorMessage > public struct SensorCadence { >> "statusMinInterval"

In this is use below function before appending data
    public var statusMinInterval: TimeInterval {
        return pow(2.0, Double(statusMinIntervalValue)) / 1000.0
    }
Due to this found that the prepared payload is not expected and in between wrong bytes data appending with this interval data bytes. as it's not in UInt8 byte type.

We found a solution for the same by replacing this wrong bytes data format with a directly bonded statusMinInterval value (UInt8 byte type) and preparing the final payload

```
    internal var data: Data {
        var data = Data([(fastCadencePeriodDivisor << 1) | statusTriggerType])
        **Existing: data += statusTriggerDelta.data + statusMinInterval
        Solution: data += statusTriggerDelta.data + statusMinIntervalValue**
        data += fastCadenceLow.data + fastCadenceHigh.data
        return data
    }
```

@philips77 
Can you please take a look at this and after review add it to upcoming releases. Also, suggest is there any use case for the "statusMinInterval" conversion function as this function return is inappropriate bytes format?

philips77, 2022-06-10:
Hi, I'll look into it next week. Thanks for reporting!

philips77, 2022-06-15:
It will be released in the upcoming version. Thank you again for the investigation.

closed 2022-06-15 by philips77