nordicsemi/IOS-nRF-Mesh-Library #447: SecKeyGeneratePair is deprecated in iOS 15 and should be replaced with SecKeyCreateRandomKey

justinmeiners, 2022-08-31:
**Describe the bug**
Apple has deprecated `SecKeyGeneratedPair`. This is used in `Crypto/Crypto.swift`.

The replacement is called `SecKeyCreateRandomKey` and it is fairly easy to make a replacement. This replacement is supported in iOS 10.0+.

**Expected behavior**

**Platform details:**
 - Device: [e.g. iPhone6]
 - OS: iOS 15.0+
 - Library Version: This is in the current main and has been there since at least late 2.x versions.

**Logs / Screenshots**
Add any logs or screenshots that can help us debug the issue faster

See https://developer.apple.com/documentation/security/1395339-seckeygeneratepair.
and https://developer.apple.com/documentation/security/1823694-seckeycreaterandomkey.

philips77, 2022-09-01:
Thank you! I'll fix it when I finally have time to get back to mesh!

philips77, 2023-05-12:
It took some time, but the issue is fixed. Thanks again for pointing it out.

PR #501 (2023-05-12): Migration from SecKeyGeneratePair to SecKeyCreateRandomKey
This PR fixes #447.
closed 2023-05-12 by philips77