nordicsemi/IOS-nRF-Mesh-Library #456: Mesh state update callbacks

daretobeorjan, 2022-10-12:
Would it be possible to add some kind of delegate or other way to signal that the mesh state has changed? The Android SDK has a onNetworkUpdated method that passes the whole MeshNetwork whenever anything in it changes.

A possible usecase would be to trigger a backup of the mesh state when needed, or a way to trigger an UI updated to reload the list of nodes.

Or, have I missed a feature available somewhere that would enable us to do the above? If not, I'm willing to implement it, given some design guidelines of how we actually want to build such a feature.

philips77, 2022-10-12:
Is it only about getting a callback when a mesh network configuration (nodes, keys, groups, scenes, ...) changes, not on each message, right?

daretobeorjan, 2022-10-12:
At the moment, for us it would be some of the more basic properties that are exported in the mesh configuration database blob. Ie addition/deletion of objects (nodes, keys, groups, like you mentioned) and some "sub-properties" of those objects (publication/subscription for node models, names, etc).

If going all in, I would say a callback on anything that is exported, at which point we would also have things like the exclusion list, heartbeat, etc. But that's beyond our usecase, so would only be if the solution needs to be "complete".