nordicsemi/IOS-nRF-Mesh-Library #457: How to get the unicast address of the connected proxy node?

FluffyBunniesTasteTheBest, 2022-10-21:
Our app allows the user to en-/disable a node's proxy and the relay features. To prevent the user from ending up with an unusable mesh, the app prevents disabling those features on the connected proxy node. 

Question now is, how do you get the unicast address of the proxy node the app is connected to?

On Android, a `ProxyConfigAddAddressToFilter` message is sent, which works well.
But when sending an `AddAddressesToFilter` message on iOS, the proxy node's address isn't forwarded to `ProxyFilterDelegate`. Is there an alternative to get the connected proxy node's address on iOS?

philips77, 2022-10-21:
I think sending a Health message to FFFF with TTL=0 and ignoring the response may be a solution?
The Health Server model uses Application Keys, but they need to be bound to the model.

I'll have a look should we add the source address to the delegate.

FluffyBunniesTasteTheBest, 2022-11-02:
@philips77  Thanks for your advise. Broadcasting a health message with TTL=0 works well.

closed 2023-11-21 by philips77