nordicsemi/IOS-nRF-Mesh-Library #459: Node's Equatable implementation is wrong

bspinner, 2022-12-22:
Hello,

we stumbled upon an issue because Node's implementation of the Equatable protocol isn't according to Apple's spec.

https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/403433c0aff5f64314fda261e90f1052c9e7853a/nRFMeshProvision/Classes/Mesh%20Model/Node.swift#L904

> Equality implies substitutability—any two instances that compare equally can be used interchangeably in any code that depends on their values. To maintain substitutability, the == operator should take into account all visible aspects of an Equatable type. Exposing nonvalue aspects of Equatable types other than class identity is discouraged, and any that are exposed should be explicitly pointed out in documentation.

In our case, we get a wrong result, because `isConfigCompleted` isn't taken into the equation.

I didn't have the time yet to look deeper into it, i.e. to check if the library requires current behaviour so additional changes would be neccessary.

philips77, 2022-12-22:
Hello,
Do you clone the Nodes in your app? Why do you have multiple instances of a Node with the same UUID? 
Indeed, we assumed that apps will just use the instances available from the app and will not remember old Nodes in case they were reprovisioned. How did you stumble upon this issue?

philips77, 2023-01-11:
I'm resuming work on this library now, so should address this issue (and all others).

PR #563 (2023-10-11): [Breaking] Fixed Equatable implementation
This PR fixes #459.

As `==` is used to detect changes in objects, e.g. in SwiftUI, the new Equatable implementations compare all parameters of objects, not just IDs. This is potentially a breaking change.
closed 2023-10-11 by philips77