nordicsemi/IOS-nRF-Mesh-Library #469: Heartbeat Message doesn't gets populated

DevHachi, 2023-02-01:
I got a question. I got my device as far that I can publish the heartbeat to my local Node, aka the Smartphone, but can't have any callback or any option to get Informed from the library about the heartbeat.

To understand why i would like to have this, is that with the heartbeat I could detect really easy if a device is reachable or not.

Would appreciate if you could help me with this.

philips77, 2023-02-01:
Hello,
To set up Heartbeats you need to do the following:
1. On the target Node set Heartbeat Publication, giving number of publications and interval, or an event. Set the destination to be the phone.
2. On the Phone's node set Heartbeat subscription, setting the sender address to the Node's address and some timeout.
3. You should be able to refresh and see the count there.

DevHachi, 2023-02-01:
So far I do understand this, but as far as i understand pub sub, shouldn't it be that I don't have to do a refresh manually but get informed like when sending a GenericOnOffSetAcknowledged?

Cause like this i would have to start a async timer that that will check the subscription in given period if the device is still there.

But as you mention the heartbeat subscription I saw that it is a single object. What happens if u have multiple devices with heartbeat subscription to the local node?

philips77, 2023-02-01:
This is not supported by the spec. Spec allows a single publication and a single subscription. Yes, that's weird. I may add multi-subscription in our libraries and ignore the spec, as that would be much more useful. I remember I was thinking about it when writing heartbeats, but then I abandoned the idea, perhaps due to no time.

Regarding a callback for received heartbeats... don't you get the global callback with heartbeat messages? I need to look into the code again and check what's there.

DevHachi, 2023-02-01:
The heartbeat messages are not populated through the MeshNetworkDelegate if u mean that. As far as I could follow the footstep through your library the heartbeat message only update the count of the subscription and thats it.

ok if the pub sub is suppose to work like that, then leave it for now as the spec says. I don't want to break that.

philips77, 2023-02-01:
I would not break it. It would add additional features on top of existing solution (without the need to subscribe the node to heartbeats, but using some new API instead).