nordicsemi/IOS-nRF-Mesh-Library #472: Unsolicited Messages are not processed

DevHachi, 2023-02-06:
When a publication and subscription is set to for example the battery status, the message is not processed but instead ignored in the Access PDU layer, because it does not find the corresponding request.

From what I understand how the pub-sub should work is, that it should also process unsolicited messages as well as requested Messages (messages like Acknowledged Gets)

philips77, 2023-02-06:
If you have created a client model with the status message type, for example like that:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/1bad249d615fc72e1fb2c15ee8abaeec631858fc/Example/Source/Mesh%20Network/GenericOnOffClientDelegate.swift#L56-L61
the status message should be delivered both to the `MeshNetworkDelegate` and your model delegate.
Mind, that in order to get statuses in the model delegate it need to be bound to the Application Key. If the statuses are sent to a group address, the model needs also to be subscribed to that group.

philips77, 2023-02-06:
If you don't have the model delegate you'll be getting an `UnknownMessage` to the `MeshNetworkDelegate`.
Remember, that you need to make sure the destination address of the status message is passed by the Proxy filter.

DevHachi, 2023-02-06:
Sorry was too early with an assumption, but the error lied with the access pdu package I got from a test hardware not implementing the battery status as the specification sepcified it. So it can be closed, works as expected.

closed 2023-02-06 by DevHachi