nordicsemi/IOS-nRF-Mesh-Library #473: Show nodes length and searching for node

Hung6129, 2023-02-16:
Hi @philips77 
I want to add a search bar for searching the node when the list is up to 100 or 200 node

And showing the node length on top on the network tab

Is look like this

[image]

Can you help me do the searching one my team really need this feature
Or you can give me some document or a way to do it that could also help me a lot

I kinda new to swift, so hope you can check it out
Thanks you

philips77, 2023-02-21:
That's a good idea. I'll try to implement it, but can't promise any date.

philips77, 2023-02-24:
Sure! Thanks! We like PRs.

philips77, 2023-05-12:
The search bar was added with #493.

closed 2023-05-12 by philips77