nordicsemi/IOS-nRF-Mesh-Library #474: Question about element 0 not bound to key

icarus31, 2023-02-19:
Hi,

I just migrated to version 3.2.0 and noticed that when using our vendor messages during the commissioning we see the last 4 lines in the log below where it says `Element 0 not bound to key` for our server and setup server.

I don't understand that I am having that. It looks like the `ConfigModelAppBind` messages for our vendor models worked and all interactions with my node are working as expected.

This output comes from the AccessLayer, but don't understand why it goes to the "else" and log that.

Any input to where I should look to solve that?

Thanks

Log

```
[Model] Sending MyMessageConfigGet(parameters: Optional(2 bytes)) from: Element 0 (0x0E01), to: 0E18
[Access] Sending Access PDU (opcode: 0xC95900, parameters: 0x0200)
[UpperTransport] Sending Upper Transport PDU (encrypted data: 0x340F83F798, transMic: 0x3E506EA5) encrypted using key: Default Key (index: 0)
[LowerTransport] Sending Access Message (akf: 1, aid: 0x1E, szmic: 0, data: 0x340F83F7983E506EA5)
[Network] Sending Network PDU (ivi: 0, nid: 0x08, ctl: 0, ttl: 2, seq: 2214, src: 0E01, dst: 0E18, transportPdu: 0x20B16E2C3790E3B14534, netMic: 0xC55F8E51) encrypted using Default Key (index: 0)
[Bearer] -> 0x00085EF530AB1B1312B420B16E2C3790E3B14534C55F8E51
[Bearer] <- 0x0008FC22F89D04BB297CF2C7232544CB3D979FAB5B7752E90951CA
[Network] Network PDU (ivi: 0, nid: 0x08, ctl: 0, ttl: 127, seq: 2111, src: 0E18, dst: 0E01, transportPdu: 0xF2C7232544CB3D979FAB5B7752, netMic: 0xE90951CA) received
[LowerTransport] Access Message (akf: 1, aid: 0x1E, szmic: 0, data: 0xB6598DC7D758DE7267DA660C) received (decrypted using key: Default Key (index: 0))
[UpperTransport] Upper Transport PDU (encrypted data: 0xB6598DC7D758DE72, transMic: 0x67DA660C) received
[Access] Response Access PDU (opcode: 0xCB5900, parameters: 0x0002005100) received (decrypted using key: Default Key (index: 0))
[Model] MyMessageConfigStatus(statusCode: Success, isSuccess: true, message: "", parameters: nil, requestType: 2) received from: 0E18, to: 0E01
[Model] Local My Server model on Element 0 (0x0E01) not bound to key: Default Key (index: 0)
[Model] MyMessageConfigStatus(statusCode: Success, isSuccess: true, message: "", parameters: nil, requestType: 2) already decoded as MyMessageConfigStatus(statusCode: Success, isSuccess: true, message: "", parameters: nil, requestType: 2)
[Model] Local My Setup Server model on Element 0 (0x0E01) not bound to key: Default Key (index: 0)
```

philips77, 2023-02-21:
Hi,

This is a very good question. Allow me to answer.

### Just like the spec wants it to be

According to Bluetooth Mesh specification mesh messages are exchanged between Models. In order to publish or receive a message a model has to be bound to an Application Key. Otherwise, the Model can't decode the message even if it supports received Opcode.
In theory then, in order to send *Generic OnOff Set* message, one would have to:
1. Have Generic OnOff Client model
2. Bind that model to an Application Key using *Config Model App Bind*
3. Set publication suing *Config Model Publication Set*
4. Use publication on event, e.g. button click.

Such configured Model can also receive *Generic OnOff Status* messages. The publication can be customized locally or remotely from a different Node that has the Device Key of the Node that is publishing.

All the above is also supported in the iOS nRF Mesh library. In order to be able to receive some messages you need to define a `ModelDelegate` which specifies which message types (Opcodes) it can received (binds an Opcode to a message type). An example can be find in the nRF Mesh app:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/1bad249d615fc72e1fb2c15ee8abaeec631858fc/Example/Source/Mesh%20Network/GenericOnOffClientDelegate.swift#L57-L59

That model delegate can publish *Generic OnOff Set Unacknowledged* messages:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/1bad249d615fc72e1fb2c15ee8abaeec631858fc/Example/Source/Mesh%20Network/GenericOnOffClientDelegate.swift#L39-L47

and does it whenever a state changes:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/1bad249d615fc72e1fb2c15ee8abaeec631858fc/Example/Source/Mesh%20Network/GenericOnOffClientDelegate.swift#L49-L54

> In the AccessLayer's handle function, I noticed the localNode is assigned to localProvisioner and not the node that is in commissioning process. So, I guess it is normal to see those message because the provisioner did not get those models bound to the app keys. Now, should the provisioner has those models bound to the app key? If so, how?

To bind a local model to an app key you have to sent *Config Model App Bind* message, just like you do to a Model on a remote Node. To make it a bit easier, you may use:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/1bad249d615fc72e1fb2c15ee8abaeec631858fc/nRFMeshProvision/Classes/MeshNetworkManager.swift#L643-L654

When a `ModelDelegate` is bound to an Application Key, the callbacks in the model delegate will get called whenever a supported message is received:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/1bad249d615fc72e1fb2c15ee8abaeec631858fc/Example/Source/Mesh%20Network/GenericOnOffClientDelegate.swift#L63-L80

The above is "the proper way how to send and receive messages according to the specification".

### To make life easier

As we're on a mobile device, which may need to send different messages each time (e.g, to turn different lights on, set different colors, etc), the library allows some simplifications. You may send ANY message just by calling
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/1bad249d615fc72e1fb2c15ee8abaeec631858fc/nRFMeshProvision/Classes/MeshNetworkManager.swift#L392-L395

without the need to bind the Application Key, set publication and publish from the model. You, actually, don't need to have a `ModelDelegate` to send messages.

> Note: The library requires to set `localElements` to something, but it can be an empty list `[]`.

For example, this approach is used on Groups tab, or from the screen of supported models in nRF Mesh app. Messages from those screens are sent either to a group, or to the opened Model without any publication.

In order to receive messages, however, you need to specify at least one `ModelDelegate` and declare supported types. That way you will get the message parsed to your type instead of `UnknownMessage`. But you don't have to bind that model delegate to an app key. The same message will be delivered to the [`MeshNetworkDelegate`](https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/main/nRFMeshProvision/Classes/MeshNetworkDelegate.swift). Indeed, in that case you will get a log warning, that the model, which declared support for this message, is not bound to the Application Key used to encrypt the message. But you can ignore it.

### Summary

In my opinion it is very nice to have an app designed according to the specification, with messages sent from models and statuses handled there. It is easier to understand, logically separated, and can be read or configured from a remote Node (if required). But that's optional in the library.It's also more difficult. Sending config messages to local node is asynchronous and can be done just once, e.g. when the app is first time configured. The configuration can be exported as well and stored in the cloud. The only thing that is hardcoded is the type of the message being published, not the publication configuration. In the example above the *Generic OnOff Set Unacknowledged* is hardcoded, while the app key, destination address, TTL, etc, are configurable.

It's up to you.

icarus31, 2023-02-22:
Hi @philips77,

I will need to digest what you wrote but it looks like the behavior I see seems normal.
Big thanks

closed 2023-02-23 by icarus31