nordicsemi/IOS-nRF-Mesh-Library #476: Strong reference cycle problem between MeshNetworkManager and NetworkManager

Ong9uJ, 2023-03-07:
`MeshNetworkManager` does not release from memory because of strong reference cycle.

If i create multiple MeshNetworkManager, none of them are released from memory.

You might need to change the reference between NetworkManager and MeshNetworkManager to a weak reference or another alternative.

philips77, 2023-03-14:
Thank you for creating the issue. I'll look into it soon.

philips77, 2023-06-01:
In #503 a `MeshNetwrokManager.clear()` methods was added, which sets the `networkManager` to `nil`, effectively breaking the cycle. This is a simple workaround for now. The proper fix removing the reference cycle will come shortly.

Ong9uJ, 2023-06-01:
Thanks very much!!!!

philips77, 2023-06-01:
Hi, could you check if the PR fixes the issue? I tried with nRF Mesh app and at least it doesn't break anything.

Ong9uJ, 2023-06-01:
It fixies my problem without any breaking. thank you!!

PR #516 (2023-06-01): Bug fix: Fixed cycling reference loop
This PR fixes #476.
closed 2023-06-01 by philips77