nordicsemi/IOS-nRF-Mesh-Library #477: Update Network and Application key at start of KeyUpdate

eanderlind, 2023-03-19:
Is there a way for application to create updates of NetworkKey and ApplicationKey for use in KeyUpdate procedure?
The api in 
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/78b532d500adecfd8fe624497217cfffb39f8c45/nRFMeshProvision/Classes/Mesh%20API/MeshNetwork%2BKeys.swift#L74

recreates a key at same index, while we instead want to invoke the (internal access) didSet  on the  existing key. 
The same api also limits binding to the first Netkey, however it looks like once we have an Appkey we can rebind it using
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/78b532d500adecfd8fe624497217cfffb39f8c45/nRFMeshProvision/Classes/Mesh%20API/ApplicationKey%2BNetworkKey.swift#L40

Hopefully just didn't find correct api.

philips77, 2023-03-20:
Hi,
Yes, it is possible. For that you have to send [`ConfigNetKeyUpdate`](https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/main/nRFMeshProvision/Classes/Mesh%20Messages/Foundation/Configuration/ConfigNetKeyUpdate.swift) or [`ConfigAppKeyUpdate`](https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/main/nRFMeshProvision/Classes/Mesh%20Messages/Foundation/Configuration/ConfigAppKeyUpdate.swift) message to [local Node](https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/78b532d500adecfd8fe624497217cfffb39f8c45/nRFMeshProvision/Classes/MeshNetworkManager.swift#L643, just like to any other Node.

The message will be handled here:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/78b532d500adecfd8fe624497217cfffb39f8c45/nRFMeshProvision/Classes/Layers/Foundation%20Layer/ConfigurationServerHandler.swift#L139-L145
Setting the key in line 141 does the following:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/78b532d500adecfd8fe624497217cfffb39f8c45/nRFMeshProvision/Classes/Mesh%20Model/NetworkKey.swift#L94-L104
The `update(networkKeyWithIndex` sets the `updated` flag in the Node's key list:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/78b532d500adecfd8fe624497217cfffb39f8c45/nRFMeshProvision/Classes/Mesh%20Model/Node.swift#L748-L756

philips77, 2023-03-20:
You may find some info here: https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/pull/314

eanderlind, 2023-03-20:
Thanks - that explains the 2nd half of the process. The remaining part is how to instantiate the two messages with the new key. Looks like only way is to generate a newkey bytestring and then hand craft a msg using the ConfigNetKeyUpdate(parameters:Data) initializer.
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/78b532d500adecfd8fe624497217cfffb39f8c45/nRFMeshProvision/Classes/Mesh%20Messages/Foundation/Configuration/ConfigNetKeyUpdate.swift#L50

Not intuitive, but we will try it out... Thanks. Closing the issue since there is a workaround.

philips77, 2023-03-21:
Great that you pointed it out. That needs improvement. I'll fix it in the next (big!) release.

PR #517 (2023-06-01): API improvements for Key Refresh Procedure
This PR fixes #477.

The API of `ConfigNetKeyUpdate` and `ConfigAppKeyUpdate` was change to allow passing the new key value.

Use the 2 messages to initiate Key Refresh Procedure on a node. The messages can also be sent to the local node.

This PR also adds of documentation for `Config[Type]Key[Action]` messages.
closed 2023-06-01 by philips77