nordicsemi/IOS-nRF-Mesh-Library #478: Annotation of @Objc for Xamarin iOS

Pallav-sculptsoft, 2023-03-20:
**Is your feature request related to a problem? Please describe.**
Need to convert in Xamarin Binding library for iOS.

**Describe the solution you'd like**
Annotation with @Objc on needed classes, methods and objects to access with binding library.

**Describe alternatives you've considered**
create framework using Objective-C.

philips77, 2023-03-20:
Please, prepare a PR. I have no experience in Xamarin and O-C is also.. hm.. let's say old. Adding @objc seems reasonable, but I don't have an app to set up. Also, some parts of the library use e.g. `OptionSet`, which is not supported in Objective-C, so I don't know how much compatible can we make it.

Pallav-sculptsoft, 2023-03-21:
Thank you @philips77 for reply,

But now we can decided to create our own now will contact with you if we can't do that things.

closed 2023-03-21 by Pallav-sculptsoft