nordicsemi/IOS-nRF-Mesh-Library #479: Re-adding the same device (same UUID) ends up with it being provisioned but not saved

philips77, 2023-03-21:
I have a device A advertising a UUID X. I provision it. Then I erase the memory and reprovision it without first removing the old instance from the app.
The provisioning succeeds (the new device gets all the keys, etc), but saving the new instance in the library fails due to the same UUID already existing. I can't reset the device as the lib lost the Device key.

The UUID check should be done BEFORE provisioning, not after.

Hung6129, 2023-03-28:
In my opinion
I think when this happen
We stop the provision process and show a pop up to tell user to remove that node manually
Or we detect it and remove it then we start the provision process

philips77, 2023-03-29:
The popup should show up before provisioning, so user can remove it manually.

philips77, 2023-06-01:
This was fixed by #512.

closed 2023-06-01 by philips77