nordicsemi/IOS-nRF-Mesh-Library #490: Awesome automation function love it

Hung6129, 2023-04-22:
I have a question about add auto publish function

Do you think it is a good idea to do an automation publish?
Does this function strongly effect or make meshnetwork unstable?

I want do something like this:
Same with automation subscribe, publish automation after choosing group, we add an other tableview for picking these parameter

    @IBOutlet weak var friendshipCredentialsFlagSwitch: UISwitch!
    @IBOutlet weak var ttlLabel: UILabel!
    @IBOutlet weak var periodSlider: UISlider!
    @IBOutlet weak var periodLabel: UILabel!
    @IBOutlet weak var retransmitCountSlider: UISlider!
    @IBOutlet weak var retransmitCountLabel: UILabel!
    @IBOutlet weak var retransmitIntervalSlider: UISlider!
    @IBOutlet weak var retransmitIntervalLabel: UILabel!

Then we move to choose model to publish
Then hit start

Thanks

philips77, 2023-04-22:
Is it common to have many models configured to publish with the same settings?

Hung6129, 2023-04-22:
The reason why I asked about the stability of the mesh network is that in our case, we have a warehouse with around 700 lights
To manage all of these lights, we publish generic OnOff and Lightness models to all of them
After a certain amount of time (1 minute in our case), the lights return their status to our dashboard.

Regarding the commonality of the issue, I think it depends on the specific case that you are facing. Whether it's about management or just a simple setup can vary greatly.

closed 2023-05-09 by Hung6129