nordicsemi/IOS-nRF-Mesh-Library #491: seqNum and IvIndex

korea1501541730, 2023-04-23:
** iPhone seqNum==0XFFFFFF,but InIndex not increase,sending message to device fail**

logs:
[Network] Sending Network PDU (ivi: 0, nid: 0x31, ctl: 0, ttl: 5, seq:1677215, src: 199A, dst: 0003, transportPdu: 0xA14B62672131BACDC64066FD, netMic: 0x6B0CC781) encrypted using Primary Network Key (index: 0)

R:How to restore it?

philips77, 2023-04-23:
The Node that is at last 50% of available Seq space should initiate IV Index Update procedure. When the IV Index increases by 1 the Seq of this mode resets to 0.

korea1501541730, 2023-04-24:
Seq>0x7FFFFF Sending Network PDU Success, the IV Index  not increases. have really seq number or (Seq>0x7FFFFF)  random num  cause the IV Index  increases by 1?

2023-04-24 13:55:26.135643+0800 Smart[16723:31117493] [Network] Sending Network PDU (ivi: 0, nid: 0x73, ctl: 0, ttl: 5, seq: 8389103, src: 199A, dst: 0007, transportPdu: 0xA08CF4D02EF37C0C87ACD53C0C, netMic: 0x05BDC71A) encrypted using Primary Network Key (index: 0)
2023-04-24 13:56:18.049021+0800 Smart[16723:31117845] [Network] Sending Network PDU (ivi: 0, nid: 0x73, ctl: 0, ttl: 5, seq: 8389104, src: 199A, dst: 0008, transportPdu: 0x6EE9BDB2F3262EE19C5FDFBA09, netMic: 0x0F1B7CB8) encrypted using Primary Network Key (index: 0)
2023-04-24 13:56:18.586209+0800 Smart[16723:31117532] [Network] Sending Network PDU (ivi: 0, nid: 0x73, ctl: 0, ttl: 5, seq: 8389105, src: 199A, dst: 0009, transportPdu: 0x26BD491CE2ADA51D2B4C3A7685, netMic: 0xD14E2384) encrypted using Primary Network Key (index: 0)
2023-04-24 13:56:18.972138+0800 Smart[16723:31117532] [Network] Sending Network PDU (ivi: 0, nid: 0x73, ctl: 0, ttl: 5, seq: 8389106, src: 199A, dst: 0009, transportPdu: 0xFBC9F41FBCC1ADD4875CDAE10B, netMic: 0xBA2AB71E) encrypted using Primary Network Key (index: 0)
2023-04-24 13:56:19.590971+0800 Smart[16723:31117845] [Network] Sending Network PDU (ivi: 0, nid: 0x73, ctl: 0, ttl: 5, seq: 8389107, src: 199A, dst: 0008, transportPdu: 0x7F64EAF963607E64D7C263F4E9, netMic: 0xB99E5785) encrypted using Primary Network Key (index: 0)
2023-04-24 13:59:27.670953+0800 Smart[16723:31119853] [Network] Sending Network PDU (ivi: 0, nid: 0x73, ctl: 0, ttl: 5, seq: 8389108, src: 199A, dst: 0008, transportPdu: 0x273F8D8D8FC534C7DE9354B328, netMic: 0x920BBCB3) encrypted using Primary Network Key (index: 0)
2023-04-24 13:59:32.367499+0800 Smart[16723:31119853] [Network] Sending Network PDU (ivi: 0, nid: 0x73, ctl: 0, ttl: 5, seq: 8389109, src: 199A, dst: 0009, transportPdu: 0xB31945E345A4A1A5E2335D8A25, netMic: 0xF6CEF2B2) encrypted using Primary Network Key (index: 0)
2023-04-24 13:59:33.637471+0800 Smart[16723:31118206] [Network] Sending Network PDU (ivi: 0, nid: 0x73, ctl: 0, ttl: 5, seq: 8389110, src: 199A, dst: 0009, transportPdu: 0x96FF575B8F64455333A6CC810A, netMic: 0x23E96EAE) encrypted using Primary Network Key (index: 0)
2023-04-24 13:59:34.791181+0800 Smart[16723:31118204] [Network] Sending Network PDU (ivi: 0, nid: 0x73, ctl: 0, ttl: 5, seq: 8389111, src: 199A, dst: 0008, transportPdu: 0x33D8CE67815FEA107420FB7E3E, netMic: 0xA1D4C615) encrypted using Primary Network Key (index: 0)
2023-04-24 13:59:35.294074+0800 Smart[16723:31118206] [Network] Sending Network PDU (ivi: 0, nid: 0x73, ctl: 0, ttl: 5, seq: 8389112, src: 199A, dst: 0008, transportPdu: 0x9BA1EF0491F46C57116E1A5C54, netMic: 0x9E28464F) encrypted using Primary Network Key (index: 0)
2023-04-24 13:59:36.278687+0800 Smart[16723:31119851] [Network] Sending Network PDU (ivi: 0, nid: 0x73, ctl: 0, ttl: 5, seq: 8389113, src: 199A, dst: 0008, transportPdu: 0x2B794FA8FE184A79D3AF6E3EC9, netMic: 0x5ABA8EBE) encrypted using Primary Network Key (index: 0)
2023-04-24 13:59:36.712829+0800 Smart[16723:31119851] [Network] Sending Network PDU (ivi: 0, nid: 0x73, ctl: 0, ttl: 5, seq: 8389114, src: 199A, dst: 0008, transportPdu: 0x8DAFA7D104B62804CDEB239895, netMic: 0xE69AD285) encrypted using Primary Network Key (index: 0)
2023-04-24 13:59:37.117241+0800 Smart[16723:31118206] [Network] Sending Network PDU (ivi: 0, nid: 0x73, ctl: 0, ttl: 5, seq: 8389115, src: 199A, dst: 0008, transportPdu: 0x2873EC06C69659E6688F4DCA53, netMic: 0xDF88135F) encrypted using Primary Network Key (index: 0)
2023-04-24 13:59:37.535415+0800 Smart[16723:31118206] [Network] Sending Network PDU (ivi: 0, nid: 0x73, ctl: 0, ttl: 5, seq: 8389116, src: 199A, dst: 0008, transportPdu: 0xCF717E1C94CEEB6060D7E8A445, netMic: 0x7C3BEC2E) encrypted using Primary Network Key (index: 0)
2023-04-24 13:59:37.982453+0800 Smart[16723:31118204] [Network] Sending Network PDU (ivi: 0, nid: 0x73, ctl: 0, ttl: 5, seq: 8389117, src: 199A, dst: 0008, transportPdu: 0x80D3BD4204F0C969ED99F1CFC4, netMic: 0x140DF14C) encrypted using Primary Network Key (index: 0)
2023-04-24 13:59:38.423654+0800 Smart[16723:31118206] [Network] Sending Network PDU (ivi: 0, nid: 0x73, ctl: 0, ttl: 5, seq: 8389118, src: 199A, dst: 0008, transportPdu: 0xABCE6110BE2CDC1ADDFE4FE806, netMic: 0xBE6E97C0) encrypted using Primary Network Key (index: 0)
2023-04-24 13:59:39.173779+0800 Smart[16723:31118204] [Network] Sending Network PDU (ivi: 0, nid: 0x73, ctl: 0, ttl: 5, seq: 8389119, src: 199A, dst: 0008, transportPdu: 0xF5D7F66668C9DE4A3F3FB2ADB2, netMic: 0x8AABA0DE) encrypted using Primary Network Key (index: 0)
2023-04-24 13:59:40.486954+0800 Smart[16723:31119851] [Network] Sending Network PDU (ivi: 0, nid: 0x73, ctl: 0, ttl: 5, seq: 8389120, src: 199A, dst: 0008, transportPdu: 0xC0910217BA0E607C2BE81860B8, netMic: 0xDD7FA1EA) encrypted using Primary Network Key (index: 0)
2023-04-24 13:59:40.892426+0800 Smart[16723:31119850] [Network] Sending Network PDU (ivi: 0, nid: 0x73, ctl: 0, ttl: 5, seq: 8389121, src: 199A, dst: 0009, transportPdu: 0xAF7B568BF329588CC386F36222, netMic: 0xF01C5461) encrypted using Primary Network Key (index: 0)
2023-04-24 13:59:41.473442+0800 Smart[16723:31118206] [Network] Sending Network PDU (ivi: 0, nid: 0x73, ctl: 0, ttl: 5, seq: 8389122, src: 199A, dst: 0008, transportPdu: 0xFFD0EF811EBA4FA7274CF9728B, netMic: 0xB7567A55) encrypted using Primary Network Key (index: 0)

philips77, 2023-04-24:
As stated in the Readme, the library will not initiate IV Index Update. This has not been implemented.
In normal circumstances other Nodes will reach 50% much earlier than a mobile phone, as the assumption is that the phones are only used for configuration and rare control.

For more information, read about IV Index Update procedure in the Bluetooth Mesh specification.

closed 2023-06-01 by philips77