nordicsemi/IOS-nRF-Mesh-Library #497: App crash after clicking on the sensor server

Hung6129, 2023-05-09:
i'm using the app that i clone using the latest update code, its crash after i try to view the model Sensor Server

[image]

But when i use the nRF Mesh on App Store it still work fine and i can config ...

[image]

in xcode it pop this
[image]

philips77, 2023-05-12:
Thank you, I'll look into that next week.

philips77, 2023-05-23:
Fixed. Thank you again for the excellent bug report! I wish all were like yours!

PR #506 (2023-05-23): Bugfix: Sensor server screen fixed
This PR fixes #497.
closed 2023-05-23 by philips77