nordicsemi/IOS-nRF-Mesh-Library #507: Incorrect units of pressure.

0x0c, 2023-05-24:
**Describe the bug**
A unit of pressure data is shown as ℃.

**To Reproduce**
You can find [here](https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/main/nRFMeshProvision/Mesh%20Messages/DeviceProperty.swift#L1297)

**Expected behavior**
A unit of pressure data is shown as hPa.

**Platform details:**
 - Device: [e.g. iPhone6]
   - Any devices
 - OS: [e.g. iOS8.1]
   - Any OS
 - Library Version (or Commit)
   - HEAD of main

**Logs / Screenshots**
Add any logs or screenshots that can help us debug the issue faster

philips77, 2023-05-24:
Nice catch!

closed 2023-05-24 by philips77

0x0c, 2023-05-25:
@philips77 Thank you for your quick fix. I noticed the sensor value of pressure is notated as Pa not hPa, so we have to divide the value of the sensor by 100. Is my understanding correct?

https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/main/nRFMeshProvision/Mesh%20Messages/DeviceProperty.swift#L1163

philips77, 2023-05-25:
I have to check it. Even if, my fix from yesterday prints it as hPa. I'll add a test for that and other similar cases.

philips77, 2023-05-25:
The comment is copy-pasted from the spec for all characteristics. I suppose it will then be in Pascals. I need to fix the description, or divided there by 10.

PR #508 (2023-05-24): Parsing sensor values fixed
This PR fixes #507.

Additionally, it fixes rounding errors when converting them to String. For example, Illuminance value 167772.13 (0xFFFFFFD) was parsed as 167772.12 as it was converted to float and represented as 167772.125 in between.