nordicsemi/IOS-nRF-Mesh-Library #531: Question regarding latest breaking change

icarus31, 2023-07-01:
Hi,

With the latest breaking change done this week, how it will be possible to cancel a config message like before?

Before, sending a config message was returning a MessageHandle, where this struct has the cancel() function.

Now, it is a ConfigResponse when sending a ConfigNodedeIdentitySet, for example.

Thanks for your help and continue the good work!

philips77, 2023-07-02:
Hello,
The non-async methods return the same `MessageHandle` as before. You shouldn't need to change anything.
The new async methods do cancel when you cancel the `Task` which they run in.

For example:
```swift
let task = Task {
   let response = try await mgr.send(...)
   // ...
}
Task {
    try sleep(nanoseconds: 1000000)
    task.cancel()
}
let response = try await task.value
```

icarus31, 2023-07-03:
Hi Philips,

But that `task.cancel()` does not behave the same as the MessageHandle.cancel(), am I wrong? If not, how can I use that one?

If doable, then not too sure to understand how it can be done. Any advice?

Thanks

icarus31, 2023-07-03:
I found out that because I was using `try await mgr.send(...)` with a config message, it was using the wrong "send()" function, not the one from the `MeshNetworkManager+Callbacks`

closed 2023-07-03 by icarus31

philips77, 2023-07-03:
Even when using `try await` version, cancelling the task will under the hood call the `MessageHandler.cancel()`:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/8a20e332fe435a388dac8473f4cc677f19f0983c/nRFMeshProvision/Layers/NetworkManager.swift#L204-L207