nordicsemi/IOS-nRF-Mesh-Library #532: Question regarding Health messages

icarus31, 2023-07-03:
Hi,

You already have `ConfigMessage` and `AcknowledgeConfigMessage` protocols.

Do you think it makes sense to have an `HealthMessage` and `AcknowledgeHealthMessage` protocols?

I am asking that because I did add those above before looking at the breaking change code, but looking at `MeshNetworkManager+Callbacks` file, there is no function having message type equivalent to that new protocol.

I am having

```
public protocol HealthMessage: StaticMeshResponse {
    // No additional fields.
}

public protocol AcknowledgedHealthMessage: HealthMessage, StaticAcknowledgedMeshMessage {
    // No additional fields.
}
```

And when calling `HealthAttentionSet`, the below send function is used.

```
@discardableResult
func send(_ message: MeshMessage,
          from localElement: Element? = nil, to model: Model,
          withTtl initialTtl: UInt8? = nil) throws -> MessageHandle {
    guard let element = model.parentElement else {
        print("Error: Element does not belong to a Node")
        throw AccessError.invalidDestination
    }
    guard let firstKeyIndex = model.bind.first,
          let meshNetwork = meshNetwork,
          let applicationKey = meshNetwork.applicationKeys[firstKeyIndex] else {
        print("Error: Model is not bound to any Application Key")
        throw AccessError.modelNotBoundToAppKey
    }
    return try send(message, from: localElement, to: MeshAddress(element.unicastAddress),
                    withTtl: initialTtl, using: applicationKey)
}

```

Do you think the HealthMessage should have been implement differently to be able to use one of the function in `MeshNetworkManager+Callbacks` file?

Regards

philips77, 2023-07-03:
Hi,
In one of the recent PRs I simplified the message structure. I removed `GenericMessage`, etc. The `ConfigMessage` was left, as these messages use Device Key for encryption. Other messages, including health messages can use `StaticAcknowledgedMeshMessage`, `StaticUnacknowledgedMeshMessage` and `StaticMeshResponse`. I refactored all other messages to use only those 3 base types.

icarus31, 2023-07-03:
Got it! Thanks!

closed 2023-07-03 by icarus31