nordicsemi/IOS-nRF-Mesh-Library #538: Why "reset()" has been removed in ProxyFilter/newProxyDidConnect() function?

icarus31, 2023-07-18:
Hi,

I noticed that in April 7th, 2022 (git hash 76aec9b), you removed the call to `reset()` in ProxyFilter/newProxyDidConnect() function.

Based on Mesh Profile 1.0.1, section 6.8.1 White List Filtering, you need to send a Set Filter Type (whitelist) before adding addresses to the filter.

You had the step before April 7th, but not anymore.

I understand that you changed a bit the `newProxyDidConnect` function, but don't you think we should have a call to `reset` before calling `setup(for: localProvisioner)`?

I probably missing something and you will have the right explanation for me :-)

Thanks in advance.

philips77, 2023-07-18:
Hi, I'm on vacations right now. Let me come back to you next week.

philips77, 2023-08-28:
Hello, I'm looking at your issue now. I'll come to you in a moment.

philips77, 2023-08-28:
The method where `reset()` was called is only called after a new Proxy was discovered.
 In 6.4.1 the spec says:
> The white list filter with an empty list is the default filter type. The Proxy Client can change the filter type as well as configure the addresses in the proxy filter.

which means, that just after connection it should be set to an empty *White List* (Accept List).

However, you're right that `setup(...)` method is public and can be called from anywhere, even if the current filter type is *Black List* (Reject List). I'll fix that.

philips77, 2023-08-28:
Ok, I know why the nRF Mesh app is working correctly when switching to a different proxy.

The `ProxyViewController` is set as a delegate of the Proxy bearer. When the bearer disconnects, this method gets called:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/80407e281e9b6736ac9444c91b18b62a14958d25/Example/Source/View%20Controllers/Proxy/ProxyViewController.swift#L183-L192
which calls `clear()` method. As the bearer is closed, the `ProxyFilter` class will fail to send this method and this code will be called:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/80407e281e9b6736ac9444c91b18b62a14958d25/nRFMeshProvision/ProxyFilter.swift#L398-L412
which resets the `type` to `.inclusionList`.

However, as this is done in the app, not by the library, some implementations may keep the former type when connected to a new Proxy.

closed 2023-08-28 by philips77