nordicsemi/IOS-nRF-Mesh-Library #540: Why NetworkLayer->updateProxyFilter implementation changed?

icarus31, 2023-07-25:
Hi,

I want to make sure that when I will select a node X and send the HealthAttentionSet message, it will directly goes to node X and not through another node.

- I first need to open a connection by looking the network id (handled in NetworkConnection). 
- Once the connection is completed and the filter calls have been terminated, I send the ConfigNodeIdentitySet message to the node I want.
- Close the connection.
- Open the proxy connection using the node identity and unicast address.
- Send HealthAttentionSet
- Close the connection

To make it work, I needed to change `updateProxyFilter` back to the old code below. Without it, the second connection was not done properly (not seeing "New Proxy connected", so the message sent after was not working, got a timeout and the resend message was working after.

````
func updateProxyFilter(usingNetworkKey networkKey: NetworkKey) {
    let justConnected = proxyNetworkKey == nil
    let reconnected = networkKey == proxyNetworkKey
    
    // Keep the primary Network Key or the most recently received one from the connected
    // Proxy Server. This is to make sure (almost) that the Proxy Configuration messages
    // are sent encrypted with a key known to this Node.
    if justConnected || networkKey.isPrimary || proxyNetworkKey?.isPrimary == false {
        proxyNetworkKey = networkKey
    }
    
    if justConnected || reconnected {
        networkManager.proxy?.newProxyDidConnect()
    }
}
```

Do you think a way to avoid coming back to the old code?

Thanks

philips77, 2023-07-25:
For reference, this is the commit that introduced the change: https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/commit/b61ea2bccca0e49140eccc054a7d57937501a542

philips77, 2023-07-25:
The issue with the old code was that when you switch GATT connection to a Proxy Node that has a different Network Key (i.e. does not know the Primary Network Key) - such connection was ignored.

philips77, 2023-07-25:
Why is `ivIndexChanged` true in your case during the second connection?

icarus31, 2023-07-25:
Not sure why you are referencing to `ivIndexChanged` variable. I could not find that variable anywhere.

I guess you meant `reconnected`?

philips77, 2023-07-26:
Sorry, my bad, the current version does not have this param:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/46fe817357d2f04a75b223f808ed8ebfbaf7e471/nRFMeshProvision/Layers/Network%20Layer/NetworkLayer.swift#L374-L385

OK, so the previous version was working for you, but the current one doesn't, right?

That indicates, that you're reconnecting to a new Node with the same Network Key (so `reconnect` was `true`).
However, the same code would not work for a Node provisioned with a different Network Key.

Perhaps the 
```swift
networkManager.proxy?.newProxyDidConnect()
```
should be called every time, without even checking `justConneted`?

The main problem here is that the Network Layer is independent from the Bearer Layer. When a Bearer changes (you switch connected Proxy Node) the Network Layer only finds our by either getting a Secure Network beacon, or when the bearer gets `BearerError.bearerClosed` error. You may have noticed, that the nRF Mesh app sends a dummy message after manually disconnecting from a Proxy Node just to trigger this error, which will clear the `proxyNetworkKey` property:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/46fe817357d2f04a75b223f808ed8ebfbaf7e471/nRFMeshProvision/Layers/Network%20Layer/NetworkLayer.swift#L221-L229

philips77, 2023-07-26:
Btw, this is what you may also do. After "Close the connection." try sending a dummy Proxy message:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/46fe817357d2f04a75b223f808ed8ebfbaf7e471/Example/Source/View%20Controllers/Proxy/ProxyViewController.swift#L187-L191

closed 2023-08-28 by philips77