nordicsemi/IOS-nRF-Mesh-Library #541: Configuration command OpCode 0x03 - Publication Set

mpiffari, 2023-08-03:
Hello folks, hi @roshanrajaratnam and @philips77 ,
I'm facing a strange behaviour with OpCode 0x03 (Config Config Model Publication Set): I write these segmented commands to characteristic

```
[0:] (BLE_MESH) 02-08-2023 17:51:51.731 [Debug] [BleMeshService.cs:1166] >  BLE MESH - Setting Publish - Segment (0) Write (30): 00-3D-75-03-30-67-47-B4-C3-79-C0-CC-CD-9B-D9-BE-5C-71-1D-F3-73-6B-EE-01-D2-4A-A7-22-D6-AB
[0:] (BLE_MESH) 02-08-2023 17:51:51.746 [Debug] [BleMeshService.cs:1166] >  BLE MESH - Setting Publish - Segment (1) Write (22): 00-3D-5B-08-A9-60-0D-11-17-E4-40-39-F6-05-64-0D-A9-6B-79-88-E2-6E
```

After these writes, I received an ACK

```
[0:] (BLE_MESH) 02-08-2023 17:51:51.915 [Debug] [BleMeshService.cs:565] >  BLE MESH - Received (25): 00-3D-C7-D4-73-1C-13-41-08-EF-23-FC-CA-62-44-66-72-B0-F4-3F-DF-BB-18-3B-73
```

and a segmented configuration model publication status (OpCode = 0x80 0x19)

```
[0:] (BLE_MESH) 02-08-2023 17:51:52.076 [Debug] [BleMeshService.cs:565] >  BLE MESH - Received (30): 00-3D-86-DF-93-B9-2F-0E-0F-42-66-86-B3-A0-CC-50-4F-AE-64-4B-DD-49-52-F1-C3-3B-6C-E9-41-E9
[0:] (BLE_MESH) 02-08-2023 17:51:52.316 [Debug] [BleMeshService.cs:565] >  BLE MESH - Received (24): 00-3D-2C-6E-0A-39-72-4A-20-D6-D0-DF-47-E3-AD-A1-68-31-E9-03-1F-7E-FC-C0
```

Up to here the flow is correct and no other messages should come as BLE notification.

But this is not true: after these receptions I obtain other 3 BLE notification with other messages (that I'm not able to decode in the same way of the previous due to crypto exception): it seems that provisionee send me more information about pubblication status.
```
[0:] (BLE_MESH) 02-08-2023 17:51:52.910 [Debug] [BleMeshService.cs:565] >  BLE MESH - Received (30): 00-3D-0D-CB-38-59-99-B9-86-8D-36-34-4E-9A-48-7C-09-8B-F4-8E-77-AB-4C-81-00-AA-DE-C1-94-3C
[0:] (BLE_MESH) 02-08-2023 17:51:53.047 [Debug] [BleMeshService.cs:565] >  BLE MESH - Received (24): 00-3D-F9-AC-36-B5-05-FB-1D-E9-C4-75-CA-17-6C-27-AF-C8-F3-FB-77-D9-7E-D7
[0:] (BLE_MESH) 02-08-2023 17:51:53.896 [Debug] [BleMeshService.cs:565] >  BLE MESH - Received (30): 00-3D-3A-5F-56-B3-30-7A-90-DA-72-3D-A1-CF-56-B2-9A-44-09-CC-5B-73-44-D0-0D-D3-A6-1E-AF-4B
[0:] (BLE_MESH) 02-08-2023 17:51:53.990 [Debug] [BleMeshService.cs:565] >  BLE MESH - Received (24): 00-3D-0B-DB-2A-1A-85-C2-69-13-61-45-A4-E7-FD-3D-0B-6D-92-14-8D-98-0C-D9
[0:] (BLE_MESH) 02-08-2023 17:51:54.841 [Debug] [BleMeshService.cs:565] >  BLE MESH - Received (30): 00-3D-FC-F6-87-03-23-37-4A-F2-A6-72-C2-16-7C-F4-79-77-32-DB-35-BA-17-DD-6C-75-B8-C5-A0-29
[0:] (BLE_MESH) 02-08-2023 17:51:54.884 [Debug] [BleMeshService.cs:565] >  BLE MESH - Received (24): 00-3D-85-16-81-EE-68-2C-47-03-A0-56-07-61-7D-C2-08-54-93-94-9D-34-9D-4F
```

Do you know what could be the problem?

philips77, 2023-08-03:
You may try our new app [nRF Mesh Sniffer](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mesh.sniffer), which can scan for mesh traffic, but only over ADV bearer.

If you're directly connected using GATT Proxy to the node that you're trying to configure, you'd have to use a different proxy and snif the messages send between proxy and your node, as messages sent over GATT are not visible there.

In the app you may add the keys and it will decode messages and parse them. Worth trying.
The app is fairly new, so any feedback is greatly appreciated.

mpiffari, 2023-08-04:
Yeah, I tried NRF Mesh Sniffer and it is absolutely useful: I need more time in order to get practice with it, but it is not clear because I cannot sniff others channel than ADV and how I could do it.
A note: the app crash if I open it with Bluetooth off and I try to activate it in app.

But, did you ever face the problem of multiple segmented response (instead of a single one) with Pubblication status OpCode? It could be something related to ACK packets sent in a timed manner?
If it worths of notice: this strange behaviour happens also if I send ACK at the end of publication status reception (also if it is an unacknowledged message).

philips77, 2023-08-04:
> A note: the app crash if I open it with Bluetooth off and I try to activate it in app.

Sorry for the crash, we'll fix it in the next release.

If you want to sniff messages sent between a phone an a proxy node you need to provision another proxy node to the network, connect the phone to the new proxy and send messages to the previous one from the phone. The messages will be sent over GATT to the new proxy and relayed over ADV to the old one, so effectively you'll be sniffing the same messages, just relayed over one extra node. Perhaps there'll be more of them, as the new proxy may add some retransmissions.

Regarding your last question, you need to try to decode the messages. If you are using nRF Mesh app on iOS you may connect it to a Mac, launch a *Console* app on the Mac and enable logs for the iPhone, then enable DEBUG and INFO logs in top menu. By filtering with `system = com.nordicsemi.nRFProvision` you'll see all logs and parsed GATT messages.

philips77, 2023-08-28:
Hi,
Did you make any progress here? Can we close the issue?

mpiffari, 2023-08-29:
Hi @philips77,
I did not manage to get more information, both with _Console_ app and tricky solution of three nodes (phone + 2 nodes) in the same network.

Actually I developed a workaround that disable BLE notifications for a certain amount of time, in order to avoid spurious messages.

closed 2023-10-10 by philips77