nordicsemi/IOS-nRF-Mesh-Library #553: Export is not conforming to spec

DomiR, 2023-09-08:
**Describe the bug**
The spec dictates the schema and id to be:
```js
{
"$schema": "http://json-schema.org/draft-04/schema#",
// depends on version, this would be currently Revision: v1.0.1
"id": "https://www.bluetooth.com/specifications/specs/mesh-cdb-1-0-1-schema.json#",
}
```
So I guess we need to enable: https://developer.apple.com/documentation/foundation/jsonencoder/outputformatting/3362874-withoutescapingslashes

This is the current output of an empty network.
```
{
	"$schema": "http:\/\/json-schema.org\/draft-04\/schema#",
	"appKeys": [],
	"groups": [],
	"id": "http:\/\/www.bluetooth.com\/specifications\/assigned-numbers\/mesh-profile\/cdb-schema.json#",
	"meshName": "nRF Mesh Network",
	"meshUUID": "2C5E4DE4-1253-4769-B9FF-D3B5D6912C31",
	"netKeys": [
		{
			"index": 0,
			"key": "38D9EEBB723B877E553A5642E3B80FCE",
			"minSecurity": "secure",
			"name": "Primary Network Key",
			"phase": 0,
			"timestamp": "2023-09-08T13:51:40Z"
		}
	],
	"networkExclusions": [],
	"nodes": [
		{
			"appKeys": [],
			"cid": "004C",
			"configComplete": true,
			"crpl": "7FFF",
			"deviceKey": "D7E4861AEAD08E048631905DEE0B0241",
			"elements": [
				{
					"index": 0,
					"location": "0001",
					"models": [
						{
							"bind": [],
							"modelId": "0000",
							"subscribe": []
						},
						{
							"bind": [],
							"modelId": "0001",
							"subscribe": []
						},
						{
							"bind": [],
							"modelId": "0002",
							"subscribe": []
						},
						{
							"bind": [],
							"modelId": "0003",
							"subscribe": []
						},
						{
							"bind": [],
							"modelId": "1205",
							"subscribe": []
						},
						{
							"bind": [],
							"modelId": "1203",
							"subscribe": []
						},
						{
							"bind": [],
							"modelId": "1204",
							"subscribe": []
						},
						{
							"bind": [],
							"modelId": "1102",
							"subscribe": []
						},
						{
							"bind": [],
							"modelId": "1008",
							"subscribe": []
						},
						{
							"bind": [],
							"modelId": "1004",
							"subscribe": []
						},
						{
							"bind": [],
							"modelId": "1005",
							"subscribe": []
						},
						{
							"bind": [],
							"modelId": "1000",
							"subscribe": []
						},
						{
							"bind": [],
							"modelId": "1002",
							"subscribe": []
						},
						{
							"bind": [],
							"modelId": "1001",
							"subscribe": []
						},
						{
							"bind": [],
							"modelId": "1003",
							"subscribe": []
						},
						{
							"bind": [],
							"modelId": "00590001",
							"subscribe": []
						}
					],
					"name": "Primary Element"
				},
				{
					"index": 1,
					"location": "0002",
					"models": [
						{
							"bind": [],
							"modelId": "1000",
							"subscribe": []
						},
						{
							"bind": [],
							"modelId": "1002",
							"subscribe": []
						},
						{
							"bind": [],
							"modelId": "1001",
							"subscribe": []
						},
						{
							"bind": [],
							"modelId": "1003",
							"subscribe": []
						}
					],
					"name": "Secondary Element"
				}
			],
			"excluded": false,
			"features": {
				"friend": 2,
				"lowPower": 2,
				"proxy": 2,
				"relay": 2
			},
			"name": "iPhone",
			"netKeys": [
				{
					"index": 0,
					"updated": false
				}
			],
			"security": "secure",
			"unicastAddress": "0001",
			"UUID": "A0CD3EBA-63E0-41EF-A219-6D99DEA5876F"
		}
	],
	"partial": false,
	"provisioners": [
		{
			"allocatedGroupRange": [
				{
					"highAddress": "CC9A",
					"lowAddress": "C000"
				}
			],
			"allocatedSceneRange": [
				{
					"firstScene": "0001",
					"lastScene": "3333"
				}
			],
			"allocatedUnicastRange": [
				{
					"highAddress": "199A",
					"lowAddress": "0001"
				}
			],
			"provisionerName": "iPhone",
			"UUID": "A0CD3EBA-63E0-41EF-A219-6D99DEA5876F"
		}
	],
	"scenes": [],
	"timestamp": "2023-09-08T13:51:40Z",
	"version": "1.0.0"
}
```

philips77, 2023-09-08:
If possible, please use the version on the *main* branch. It has this fixed and is migrated to CDB 1.0.1.
I'll release version 4.0 of nRF Mesh app together with the library next week after implementing the new SAR. It's the last missing feature.

DomiR, 2023-09-08:
Same on main (3c6cde5f07929872d11542dd27441776bf697580). After applying the following patch it works as expected.

```
diff --git a/nRFMeshProvision/MeshNetworkManager.swift b/nRFMeshProvision/MeshNetworkManager.swift
index c755f850..71d0ee63 100644
--- a/nRFMeshProvision/MeshNetworkManager.swift
+++ b/nRFMeshProvision/MeshNetworkManager.swift
@@ -1204,8 +1204,9 @@ public extension MeshNetworkManager {
     func export() -> Data {
         let encoder = JSONEncoder()
         encoder.dateEncodingStrategy = .iso8601
+        encoder.outputFormatting = .withoutEscapingSlashes
         if #available(iOS 11.0, *) {
-            encoder.outputFormatting = .sortedKeys
+            encoder.outputFormatting = [.sortedKeys, .withoutEscapingSlashes]
         }
         
         return try! encoder.encode(meshData.meshNetwork)
@@ -1233,8 +1234,9 @@ public extension MeshNetworkManager {
     func export(_ configuration: ExportConfiguration) -> Data {
         let encoder = JSONEncoder()
         encoder.dateEncodingStrategy = .iso8601
+        encoder.outputFormatting = .withoutEscapingSlashes
         if #available(iOS 11.0, *) {
-            encoder.outputFormatting = .sortedKeys
+            encoder.outputFormatting = [.sortedKeys, .withoutEscapingSlashes]
         }
         
         let meshNetwork = meshData.meshNetwork?.copy(using: configuration)
```

philips77, 2023-09-08:
Wow, thanks. I didn't know about the backslashes and the app is there for years now. Boomer :)
And thanks for the patch.

PR #558 (2023-10-10): Bugfix: Exporting URIs without escaping slashes
This PR fixes #553.
closed 2023-10-10 by philips77