nordicsemi/IOS-nRF-Mesh-Library #554: Import does not allow fracSec in timestamp

DomiR, 2023-09-08:
**Describe the bug**
The spec is a bit fuzzy on this, as it does not give an example with fracSecs but I quote:

> MeshCDB 2.1.2
> The timestamp is based on UTC and follows the “date-time” format as defined by JSON Schema Draft 4 [3], which is based on Internet date/time format described in Section 5.6 of RFC 3339 [6]:
> ```
> YYYY-MM-DDThh:mm:ssZ or YYYY-MM-DDThh:mm:ss+/- timeoffset
> ```
> 

Which means we should probably allow importing timestamps that follow the json schema draft:
https://datatracker.ietf.org/doc/html/draft-fge-json-schema-validation-00#section-7.3.1.2

Which allows for optional frac secs:
https://datatracker.ietf.org/doc/html/rfc3339#section-5.6

Meaning we probably want to parse using a custom date decoder: https://stackoverflow.com/a/28016692
Encoding does not need factional seconds.

This is not really important, just "more" spec compliant. Will change our export for now.

philips77, 2023-09-08:
Yes, we should not reject json just due to extra milliseconds. We'll change it.

philips77, 2023-10-10:
Fixed. Will be released soon as 4.0.0.

PR #560 (2023-10-10): Support for importing configuration with fractional seconds
This PR fixes #554.
closed 2023-10-10 by philips77