nordicsemi/IOS-nRF-Mesh-Library #556: Why group address not added in networkExclusions?.excludedAddresses when deleted?

icarus31, 2023-10-08:
Hi,

In theory, if a group is no more used, its address should not be added in the network's excludedAddresses array?

If it is the case, then nextAvailableGroupAddress should check inside that array too, correct?

If not needed, what is the reason behind?

Regards

philips77, 2023-10-10:
Group addresses can be reused, as there's no reason why they wouldn't be.

For Unicast Address the reason is the RPL (Replay Protection List). Each node remembers the last SEQ number for any Unicast Address from which it has received any message in order to prevent reply attacks (resending the same message by an impostor). If a Unicast Address would be reused before the IV Index rolls by 2 other nodes would discard all messages from the new node until its SEQ reaches a higher value than the last one used by the old node. When IV Index rolls to by 2 since the node was removed the Unicast Address can be reassigned, as the AuthSeq consists of IV Index + SEQ. With higher IV Index the SeqAuth will always be higher than the one used by the old node.

And why the IV Index has to roll by 2? During IV Update procedure there's a time where a node can accept messages sent with the previous IV Index, as not all devices must transition to the new IV Index immediately. They have 96h to do so since the IV Update procedure begins.

closed 2023-10-10 by philips77

icarus31, 2023-10-11:
Hi @philips77 , thanks for the quick response.

So based on what you said, the node does not remember the last SEQ number for each group address message it received? I understand it can make the difference between an unicast and a group address, but both are addresses

I thought that an imposter could also resend the same group message similar to what he could do with an unicast address. That was the reason behind my question. :-)

philips77, 2023-10-24:
It is the address of the sender (unicast address) that matters. And the sender knows keys, can encrypt messages, which means is trusted. An impostor doesn't know keys. Can only sniff, collect messages and try to resend them again. But as they were already "used", they won't work. The destination address, whether it is a group, virtual, or unicast address, doesn't matter.

If a unicast address is reused for a new node it would have to start sending messages with "unique" sequence (higher than last one used by the old node), and this is not known. You'd have to send messages over and over in a loop until they start work, which may take days.