nordicsemi/IOS-nRF-Mesh-Library #568: Version 4.0.0 not sent vender mesages

DuanPhan92, 2023-10-14:
**Describe the bug**
I am using nrf mesh app on my iphone, yesterday everthing still ok. But today i can’t send any messages in vendor model.

**Platform details:**
 - Device: Iphone XS
 - OS: IOS16.6.1
 - Library Version 4.0.0

philips77, 2023-10-16:
Thank you for the bug report. I'm in process of fixing bug, so I'll make sure to test and fix it.

Do you have any logs?

philips77, 2023-10-16:
OK, I found the issue. `VendorMessage` is neither `AcknowledgedVendorMessage` nor `UnacknowledgedVendorMessage` and therefore is never sent. Will fix now!

closed 2023-10-17 by philips77

philips77, 2023-10-17:
Version 4.0.1 has been submitted to AppStore. Should be available soon.
Once again thank you for reporting the bug.

PR #570 (2023-10-17): Vendor messages fixed in the nRF Mesh app
This PR fixes #568.

Other improvements:
* As the response Op Code for a custom acknowledged vendor message is not known, it has to be provided by the user. A new text field was added, which is enabled only when *Acknowledged* switch is on.
* Values for all fields are stored and loaded next time you enter the model view.