nordicsemi/IOS-nRF-Mesh-Library #576: Unable to send a message to specified address. Another transfer in progress.

xaviergspintly, 2023-10-24:
### Where do you suspect the issue?

Other

### Version

4.0.1 (Latest)

### Describe the issue

We were using the latest 4.0.1 release in our app to try and configure devices.
We are using `AcknowledgedConfigMessage` and `MeshMessage`.
- The `AcknowledgedConfigMessage` for sending `ConfigCompositionDataGet`, `ConfigAppKeyAdd`, `ConfigModelAppBind` and others
- the `MeshMessage` for sending `RuntimeUnacknowledgedVendorMessage` as per your sample code of nRFMesh repo (isSegmented = false, security = .low)

Below is how we use the `send()` functions provided:
- For `AcknowledgedConfigMessage` we use
`try self.meshNetworkManager.send(message as! AcknowledgedConfigMessage, to: unicastAddress ?? 0, completion: { response in
switch(response) {
 case .success( _):
  print("sendMeshMessage message != nil AcknowledgedConfigMessage success")
  break
 case .failure(let error as NSError):
  print("sendMeshMessage message != nil AcknowledgedConfigMessage failure \(error.localizedDescription)")
  //retry logic with 600 msec backoff
  break
}
})`
- For `MeshMessage` i.e. `RuntimeUnacknowledgedVendorMessage` we use
`try self.meshNetworkManager.send(message, to: MeshAddress(unicastAddress ?? 0), using: appKey, completion: { response in
switch(response) {
 case .success( _):
  print("sendMeshMessage message != nil MeshMessage success")
  break
 case .failure(let error as NSError):
  print("sendMeshMessage message != nil MeshMessage failure \(error.localizedDescription)")
  //retry logic with 600 msec backoff
  break
}
})`

We send the next message only after we reach the **`.success()` of the previous message**. Success status messages of the previous message gets printed and then only new message `send()` is tried.

However we are entering the `.failure()` case of the `send()` function sometimes. And the error printed is

> Unable to send a message to specified address. Another transfer in progress.

To circumvent this we added backoff delay to retry sending the same `send()` after 600 msec and 3 retries but we still end up with the failure consistently. Nothing gets printed via the  nRFMesh logger in this time and neither is it sending anything as per the bearer prints (which sends data over bluetooth) which are not coming during this time.

The funny part is that for the same mesh device, this error comes about 8/10 times on an iPhone 12 mini but it hardly comes 2/10 times on an iPhone 11. Both the phones are on iOS 16 above and the app supports iOS 13 and above.

Like I can see `guard !outgoingMessages.contains(destination) else {}` in multiple places of the `NetworkManager.swift`. But why is it entering the `else` or the places where `AccessError.busy` is thrown? Is `outgoingMessages` not cleared or something? Or is something else the problem?

In general, why is the success rate dependant on phone model and what is the solution? Kind of need resolution on this soon to release an app.

### Relevant log output

```text
2023-10-23T16:56:28.380+05:30 Entered sendMeshMessage(),
2023-10-23T16:56:28.381+05:30 sendMeshMessage message != nil MeshMessage failure Unable to send a message to specified address. Another transfer in progress.,
2023-10-23T16:56:28.904+05:30 Entered sendMeshMessage(),
2023-10-23T16:56:28.904+05:30 sendMeshMessage message != nil MeshMessage failure Unable to send a message to specified address. Another transfer in progress.,
2023-10-23T16:56:29.004+05:30 Entered sendMeshMessage(),
2023-10-23T16:56:29.004+05:30 sendMeshMessage message != nil MeshMessage failure Unable to send a message to specified address. Another transfer in progress.,
2023-10-23T16:56:29.533+05:30 Entered sendMeshMessage(),
2023-10-23T16:56:29.534+05:30 sendMeshMessage message != nil MeshMessage failure Unable to send a message to specified address. Another transfer in progress.

2023-10-23T17:08:37.960+05:30 sendMeshMessage message != nil AcknowledgedConfigMessage success,
2023-10-23T17:08:37.961+05:30 sendMeshMessage message != nil AcknowledgedConfigMessage failure Unable to send a message to specified address. Another transfer in progress.,
2023-10-23T17:08:38.580+05:30 Entered sendMeshMessage(),
2023-10-23T17:08:38.581+05:30 sendMeshMessage message != nil AcknowledgedConfigMessage failure Unable to send a message to specified address. Another transfer in progress.,
2023-10-23T17:08:39.198+05:30 Entered sendMeshMessage(),
2023-10-23T17:08:39.198+05:30 sendMeshMessage message != nil AcknowledgedConfigMessage failure Unable to send a message to specified address. Another transfer in progress.
```

philips77, 2023-10-24:
Hello,
Ok, let's debug this together.
The error you are getting is this one:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/e8a91907650d454335e87c57fa0c5c12800a48f9/nRFMeshProvision/Layers/Access%20Layer/AccessError.swift#L78
It is thrown whenever `NetworkManager` thinks the last message to the given destination wasn't sent, for example here:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/e8a91907650d454335e87c57fa0c5c12800a48f9/nRFMeshProvision/Layers/NetworkManager.swift#L205-L208
and 3 other similar places in this file.

The assumption is that when a message gets sent this method is called:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/e8a91907650d454335e87c57fa0c5c12800a48f9/nRFMeshProvision/Layers/NetworkManager.swift#L560-573
which removes the message matching the destination.

Or, when a failure occurs, it is cleared here:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/e8a91907650d454335e87c57fa0c5c12800a48f9/nRFMeshProvision/Layers/NetworkManager.swift#L583-L588

Only after that it is possible to send new message. The completion callbacks are called AFTER the `outgoingMessages` was cleared, so it should be possible to send new message from there.

However, as the lower transport layer SAR implementation is new, there may be bugs.

Could you share logs showing when this issue happens? If you're using nRF Mesh app, you may open *Console* app (native on Mac), start logging for a connected iPhone, enable Info and Debug logs in menu and set filter for `subsystem=com.nordicsemi.nRFProvision` to show only mesh logs. That would help a lot.

xaviergspintly, 2023-10-25:
I have used the Mesh Logger functionality in our app using the nRFMesh library and the failure log is as below (there may be repetitive function prints, but it is just an iterator recursively calling it again and not the same function being called from different threads  eg. like in 2023-10-25 17:31:12.463086+0530):

`
Log to share:
2023-10-25 17:30:55.100666+0530 IosBasicConfigurationTest[1128:358457] [Logger] INFO /BleMeshManager.swift centralManagerDidUpdateState(_:) 612 stopping scan
2023-10-25 17:30:55.100970+0530 IosBasicConfigurationTest[1128:358457] [Logger] DEBUG /MeshConfigurationManager.swift didUpdateBleState(available:) 2834 didUpdateBleState: available: true
2023-10-25 17:30:55.101210+0530 IosBasicConfigurationTest[1128:358457] [Logger] INFO /BleMeshManager.swift centralManagerDidUpdateState(_:) 612 stopping scan
2023-10-25 17:30:55.101419+0530 IosBasicConfigurationTest[1128:358457] [Logger] DEBUG /MeshConfigurationManager.swift didUpdateBleState(available:) 2834 didUpdateBleState: available: true
2023-10-25 17:31:01.669040+0530 IosBasicConfigurationTest[1128:358457] [Logger] DEBUG /BleMeshManager.swift centralManager(_:didConnect:) 712 Connected to 281B6ED9-D04F-B471-C330-434B893131CE
2023-10-25 17:31:01.669322+0530 IosBasicConfigurationTest[1128:358457] [Logger] DEBUG /MeshConfigurationManager.swift didConnect(_:) 2524 didConnect: device: Optional("Target_device"), state: Optional(ConfigurationStates.PROVISIONING_STARTED)
2023-10-25 17:31:02.113324+0530 IosBasicConfigurationTest[1128:358532] [Logger] DEBUG /BleMeshManager.swift peripheral(_:didDiscoverServices:) 757 didDiscoverServices 6F400001-B5A3-F393-E0A9-E50E24DCCA9E
2023-10-25 17:31:02.114214+0530 IosBasicConfigurationTest[1128:358532] [Logger] DEBUG /BleMeshManager.swift peripheral(_:didDiscoverServices:) 757 didDiscoverServices 1827
2023-10-25 17:31:02.292808+0530 IosBasicConfigurationTest[1128:358532] [Logger] DEBUG /BleMeshManager.swift initPeripheral(peripheral:) 245 Service discovery completed for 281B6ED9-D04F-B471-C330-434B893131CE
2023-10-25 17:31:02.293166+0530 IosBasicConfigurationTest[1128:358532] [Logger] DEBUG /BleMeshManager.swift initPeripheral(peripheral:) 248 didDiscoverCharacteristicsFor: BLUETOOTH_DATA_IN
2023-10-25 17:31:02.293508+0530 IosBasicConfigurationTest[1128:358532] [Logger] DEBUG /BleMeshManager.swift initPeripheral(peripheral:) 275 didDiscoverCharacteristicsFor: MESH_PROVISIONING_UUID
2023-10-25 17:31:02.294302+0530 IosBasicConfigurationTest[1128:358532] [Logger] DEBUG /BleMeshManager.swift iterateOverNotifyUuids(_:) 316 iterateOverNotifyUuids()
2023-10-25 17:31:02.294644+0530 IosBasicConfigurationTest[1128:358532] [Logger] DEBUG /BleMeshManager.swift iterateOverNotifyUuids(_:) 327 iterateOverNotifyUuids: currentNotifyUuid: Optional("00002ADC-0000-1000-8000-00805F9B34FB")
2023-10-25 17:31:02.294920+0530 IosBasicConfigurationTest[1128:358532] [Logger] DEBUG /BleMeshManager.swift doPerformNotifyRequest(forPeripheral:to:) 536 doPerformNotifyRequest: peripheral: 281B6ED9-D04F-B471-C330-434B893131CE withUUID: 00002ADC-0000-1000-8000-00805F9B34FB
2023-10-25 17:31:02.471511+0530 IosBasicConfigurationTest[1128:358457] [Logger] DEBUG /BleMeshManager.swift peripheral(_:didUpdateNotificationStateFor:error:) 891 didUpdateNotificationStateFor: 281B6ED9-D04F-B471-C330-434B893131CE 2ADC
2023-10-25 17:31:02.471914+0530 IosBasicConfigurationTest[1128:358457] [Logger] DEBUG /BleMeshManager.swift iterateOverNotifyUuids(_:) 316 iterateOverNotifyUuids()
2023-10-25 17:31:02.472183+0530 IosBasicConfigurationTest[1128:358457] [Logger] DEBUG /BleMeshManager.swift iterateOverNotifyUuids(_:) 327 iterateOverNotifyUuids: currentNotifyUuid: Optional("6F400003-B5A3-F393-E0A9-E50E24DCCA9E")
2023-10-25 17:31:02.472447+0530 IosBasicConfigurationTest[1128:358457] [Logger] DEBUG /BleMeshManager.swift doPerformNotifyRequest(forPeripheral:to:) 536 doPerformNotifyRequest: peripheral: 281B6ED9-D04F-B471-C330-434B893131CE withUUID: 6F400003-B5A3-F393-E0A9-E50E24DCCA9E
2023-10-25 17:31:02.591902+0530 IosBasicConfigurationTest[1128:358457] [Logger] DEBUG /BleMeshManager.swift peripheral(_:didUpdateNotificationStateFor:error:) 891 didUpdateNotificationStateFor: 281B6ED9-D04F-B471-C330-434B893131CE 6F400003-B5A3-F393-E0A9-E50E24DCCA9E
2023-10-25 17:31:02.592269+0530 IosBasicConfigurationTest[1128:358457] [Logger] DEBUG /BleMeshManager.swift iterateOverNotifyUuids(_:) 316 iterateOverNotifyUuids()
2023-10-25 17:31:02.592488+0530 IosBasicConfigurationTest[1128:358457] [Logger] DEBUG /BleMeshManager.swift identifyOrCompleteSetup(_:) 333 identifyOrCompleteSetup() isIdentificationProcessRequired: false
2023-10-25 17:31:02.592844+0530 IosBasicConfigurationTest[1128:358457] [Logger] DEBUG /MeshConfigurationManager.swift didCompleteSetup(_:) 2581 didCompleteSetup: device: Optional("Target_device"), state: Optional(ConfigurationStates.PROVISIONING_STARTED)
2023-10-25 17:31:02.593980+0530 IosBasicConfigurationTest[1128:358457] [Logger] DEBUG /MeshConfigurationManager.swift didCompleteSetup(_:) 2608 didCompleteSetup: Before provisioning
2023-10-25 17:31:02.595196+0530 IosBasicConfigurationTest[1128:358457] [Logger] DEBUG /MeshConfigurationManager.swift identifyNode(_:) 860 identifyNode:serviceData 1153501018A00B00006E0000000000000000
2023-10-25 17:31:02.595573+0530 IosBasicConfigurationTest[1128:358457] [Logger] DEBUG /MeshConfigurationManager.swift identifyNode(_:) 870 provisioningManager state: Requesting Provisioning Capabilities
[... cut ...]
2023-10-25 17:31:16.056001+0530 IosBasicConfigurationTest[1128:358560] [Logger] INFO /MeshConfigurationManager.swift log(message:ofCategory:withLevel:) 168 nRF network: Sending Network PDU (ivi: 0, nid: 0x39, ctl: 0, ttl: 5, seq: 256, src: 0005, dst: 0006, transportPdu: 0xEB5EEA9857C7A67391460A7C9792EAF5, netMic: 0x7D466989) encrypted using Network Key 1 (index: 0)
2023-10-25 17:31:16.056659+0530 IosBasicConfigurationTest[1128:358560] [Logger] WARN /MeshConfigurationManager.swift log(message:ofCategory:withLevel:) 168 nRF lowerTransport: The bearer is closed.

`

anees17861, 2023-10-30:
Hi @philips77 ,

I had went through the logs along with @xaviergspintly and after observing, we found that we were not receiving acks for segmented messages sometimes. We will be looking into the firmware side of the issue as well.

We send the next config message once we get a status message for the previous config message. (From the completion in the send message function). This can cause a race condition if there is some delay in device sending the segmentation ack causing the config status message to be received first and processed.

Unfortunately it can't be handled directly on app side, since there is no indication to wether segmented message was successfully delivered and acknowledged for config messages.

> 2023-10-25 17:31:12.433983+0530 IosBasicConfigurationTest[1128:358560] [Logger] DEBUG /CustomBearer.swift didReceiveData(pdu:) 51 CustomBearer RX: 00395581C5D526B752A383F8A756D1FD1C738D5B02F66F9975EE248D5A6B

> 2023-10-25 17:31:12.435965+0530 IosBasicConfigurationTest[1128:358560] [Logger] DEBUG /CustomBearer.swift didReceiveData(pdu:) 51 CustomBearer RX: 00395581C5D526B752A383F8A756D1FD1C738D5B02F66F9975EE248D5A6B

There is another odd behaviour, is that in some of the failure logs, we see duplicate messages being received by the app. Unless that is expected, what I believe is happening that there might be a race happening on firmware side. The segmented ack might be getting notified on the characteristic but before the data is read by the mobile, it might be getting overriden by the config status ack. Hence mobile may end up reading config status ack in both the messages.

Please let us know if this analysis makes sense or is the duplicate message something expected of mesh protocol

philips77, 2023-10-30:
Hello,
> ...we found that we were not receiving acks for segmented messages sometimes. We will be looking into the firmware side of the issue as well.

I got reported this missing ACK also internally. I will look into it. I got some fw with which I should be able to reproduce the issue. I went through your log and cleaned it a bit, removed unnecessary logs, split so that messages are easier to see.

> There is another odd behaviour, is that in some of the failure logs, we see duplicate messages being received by the app.

Indeed, I noticed that as well. It may be an issue but doesn't have to. In Bluetooth Mesh there's something called *Network Transmit* parameter. It defines how many times the network layer should resend a packet to increase changes of delivery. This is exactly the same packet, with the same SEQ value. There's no need to send such packets over GATT bearer, but there are use cases for sending them over ADV bearer. In your case it may be, that the Network Transmit for the Proxy node is 2 and it also sends them over GATT. However, not all packets are duplicated, so indeed this looks like some kind of an issue.

xaviergspintly, 2023-10-30:
Also @philips77 below is how the parameters have been set up as per your nRFMesh sample code (if it is of any help):

```
meshNetworkManager.networkParameters = .basic { parameters in
            parameters.setDefaultTtl(5)
            // Configure SAR Receiver properties
            parameters.discardIncompleteSegmentedMessages(after: 10.0)
            parameters.transmitSegmentAcknowledgmentMessage(
                usingSegmentReceptionInterval: 0.06,
                multipliedByMinimumDelayIncrement: 2.5)
            parameters.retransmitSegmentAcknowledgmentMessages(
                exactly: 1, timesWhenNumberOfSegmentsIsGreaterThan: 3)
            // Configure SAR Transmitter properties
            parameters.transmitSegments(withInterval: 0.06)
            parameters.retransmitUnacknowledgedSegmentsToUnicastAddress(
                atMost: 2, timesAndWithoutProgress: 2,
                timesWithRetransmissionInterval: 0.200, andIncrement: 2.5)
            parameters.retransmitAllSegmentsToGroupAddress(exactly: 3, timesWithInterval: 0.250)
            
            // Note: The values below are different from the default ones.
            
            // Configure message configuration
            parameters.retransmitAcknowledgedMessage(after: 4.2)
            // As the interval has been increased, the timeout can be adjusted.
            // The acknowledged message will be repeated after 4.2 seconds,
            // 12.6 seconds (4.2 + 4.2 * 2), and 29.4 seconds (4.2 + 4.2 * 2 + 4.2 * 4).
            // Then, leave 10 seconds for until the incomplete message times out.
            parameters.discardAcknowledgedMessages(after: 40.0)
        }
```

closed 2023-10-30 by philips77

anees17861, 2023-11-01:
Hi @philips77 ,

We found two issues, one in our app and another in the library.

In app side. In our CBCentralManager side implementation, we were doing all operations on a serial queue asynchronously. For eg. in the didUpdateValue for characteristic callback, all the code within the function definition was wrapped in a serialQueue.async {} block. This caused the characteristic.value to be read at a later time than the time iOS actually called our didUpdateValue.

Device was sending notifications 5-6 ms apart which was not enough time for the async blocks to start. So by the time the first async block started, didUpdateValue was already triggered multiple times causing the characteristic.value to be overiden. Solution was to simply extract out the characteristic value before going into the async block.

In the library side though we observed the messages getting processed out of order. For eg, we saw messages coming in characteristic update in sequence order 7,8,9 but in library they were being processed in 8,9,7. This caused the seq=7 message to be dropped since it was not lesser than the current seq = 9.

We believe the issue is arising at the following line. The task.detached may cause things to be processed out of order. If it makes sense, maybe it can be moved to a serial dispatch queue instead. 
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/244781633222a9c64225d57a23a56b8a10d222c4/nRFMeshProvision/MeshNetworkManager.swift#L262-L264

xaviergspintly, 2023-11-01:
@philips77 the issue of receiving same messages stemmed  from our implementation where we were doing characteristic read in an async manner. App was getting notified thrice but since we were taking data for read in an async manner (to prevent from hanging in the didUpdateValueFor function which may halt the bluetooth), only the last packet was getting read thrice. Doesn't look like there was anything related to the library, since, this was before data was fed to the library (custom bearer).

We fixed that part but ran into issue #586 on which we request your assistance.

anees17861, 2023-11-01:
Moving the conversation to #586

PR #581 (2023-10-30): Bug fix: ACK was sending too late
This PR fixes #576.

When a segmented message was received, the ACK confirming the complete message was scheduled to be sent in a background queue. This caused a delay. The manager's callback was notified immediately and was able to send another message before the ACK was sent. The ACK would then be discarded by the Proxy node as having too low sequence number.

This PR removes this delay by sending ACK immediately without using the queue.