nordicsemi/IOS-nRF-Mesh-Library #586: Message sequences received out of sync

xaviergspintly, 2023-10-31:
### Where do you suspect the issue?

Issue in mesh library, e.g. packets being dropped

### Version

4.0.1 (Latest)

### Describe the issue

@philips77 our previous issue of receiving same messages was because we were doing characteristic read in an async manner. We sorted that out but we ran into the issue as shown in the log below.
Messages are received out of sequence due to which they are flagged.
See the sequence 7, 8, 9 below.

### Relevant log output

```text
2023-10-31 20:00:16.126922+0530 IosBasicConfigurationTest[1837:811187] [.Logger] DEBUG/CustomBearer.swift didReceiveData(pdu:) 51 CustomBearer RX: 003906FBE62E75F7098AFA51495AA4C96893DB590373476928
2023-10-31 20:00:16.127328+0530 IosBasicConfigurationTest[1837:811187] [.Logger] DEBUG/CustomBearer.swift didReceiveData(pdu:) 53 CustomBearer reassemble: (data: 24 bytes, messageType:.PduType.networkPdu)
2023-10-31 20:00:16.127963+0530 IosBasicConfigurationTest[1837:811187] [.Logger] DEBUG/BleMeshManager.swift peripheral(_:didUpdateValueFor:error:) 827 after async didUpdateValueFor: 281B6ED9-D04F-B471-C330-434B893131CE Characteristic: 2ADE Data: Optional("003946691F6085A26CB95D21F8849A2AE82FB67D0340AD5DF99952092818")
2023-10-31 20:00:16.128532+0530 IosBasicConfigurationTest[1837:811187] [.Logger] DEBUG/MeshConfigurationManager.swift didReceiveData(_:characteristicUUID:data:) 2743 didReceiveData: device: Optional(""), state: Optional(.ConfigurationStates.CONFIGURATION_CONNECTED), characteristicUUID: 00002ADE-0000-1000-8000-00805F9B34FB, data: 003946691F6085A26CB95D21F8849A2AE82FB67D0340AD5DF99952092818
2023-10-31 20:00:16.129022+0530 IosBasicConfigurationTest[1837:811187] [.Logger] DEBUG/CustomBearer.swift didReceiveData(pdu:) 51 CustomBearer RX: 003946691F6085A26CB95D21F8849A2AE82FB67D0340AD5DF99952092818
2023-10-31 20:00:16.129316+0530 IosBasicConfigurationTest[1837:811187] [.Logger] DEBUG/CustomBearer.swift didReceiveData(pdu:) 53 CustomBearer reassemble: (data: 29 bytes, messageType:.PduType.networkPdu)
2023-10-31 20:00:16.129871+0530 IosBasicConfigurationTest[1837:811187] [.Logger] DEBUG/BleMeshManager.swift peripheral(_:didUpdateValueFor:error:) 827 after async didUpdateValueFor: 281B6ED9-D04F-B471-C330-434B893131CE Characteristic: 2ADE Data: Optional("0039392EC9B87D6A5FC143D7DABA5F6DAA26A905C33CEF1C058A")
2023-10-31 20:00:16.130553+0530 IosBasicConfigurationTest[1837:811059] [.Logger] INFO/MeshConfigurationManager.swift log(message:ofCategory:withLevel:) 168 nRF network: Network PDU (ivi: 0, nid: 0x39, ctl: 0, ttl: 8, seq: 8, src: 0006, dst: 0007, transportPdu: 0x5D21F8849A2AE82FB67D0340AD5DF999, netMic: 0x52092818) received
2023-10-31 20:00:16.130782+0530 IosBasicConfigurationTest[1837:811187] [.Logger] DEBUG/MeshConfigurationManager.swift didReceiveData(_:characteristicUUID:data:) 2743 didReceiveData: device: Optional(""), state: Optional(.ConfigurationStates.CONFIGURATION_CONNECTED), characteristicUUID: 00002ADE-0000-1000-8000-00805F9B34FB, data: 0039392EC9B87D6A5FC143D7DABA5F6DAA26A905C33CEF1C058A
2023-10-31 20:00:16.131232+0530 IosBasicConfigurationTest[1837:811187] [.Logger] DEBUG/CustomBearer.swift didReceiveData(pdu:) 51 CustomBearer RX: 0039392EC9B87D6A5FC143D7DABA5F6DAA26A905C33CEF1C058A
2023-10-31 20:00:16.132163+0530 IosBasicConfigurationTest[1837:811187] [.Logger] DEBUG/CustomBearer.swift didReceiveData(pdu:) 53 CustomBearer reassemble: (data: 25 bytes, messageType:.PduType.networkPdu)
2023-10-31 20:00:16.132906+0530 IosBasicConfigurationTest[1837:811187] [.Logger] INFO/MeshConfigurationManager.swift log(message:ofCategory:withLevel:) 168 nRF network: Network PDU (ivi: 0, nid: 0x39, ctl: 0, ttl: 8, seq: 9, src: 0006, dst: 0007, transportPdu: 0x43D7DABA5F6DAA26A905C33C, netMic: 0xEF1C058A) received
2023-10-31 20:00:16.131719+0530 IosBasicConfigurationTest[1837:811266] [.Logger] INFO/MeshConfigurationManager.swift log(message:ofCategory:withLevel:) 168 nRF network: Network PDU (ivi: 0, nid: 0x39, ctl: 1, ttl: 8, seq: 7, src: 0006, dst: 0007, transportPdu: 0xFA51495AA4C968, netMic: 0x93DB590373476928) received
2023-10-31 20:00:16.137008+0530 IosBasicConfigurationTest[1837:811059] [.Logger] DEBUG/MeshConfigurationManager.swift log(message:ofCategory:withLevel:) 168 nRF lowerTransport: Segmented Access Message (akf: 0, szmic: 0, seqZero: 8, segO: 0, segN: 1, data: 0x0F8757545F85D36D85E26470) received (decrypted using key: Network Key 1 (index: 0))
2023-10-31 20:00:16.139622+0530 IosBasicConfigurationTest[1837:811187] [.Logger] DEBUG/MeshConfigurationManager.swift log(message:ofCategory:withLevel:) 168 nRF lowerTransport: Segmented Access Message (akf: 0, szmic: 0, seqZero: 8, segO: 1, segN: 1, data: 0x00569193974EA701) received (decrypted using key: Network Key 1 (index: 0))
2023-10-31 20:00:16.139967+0530 IosBasicConfigurationTest[1837:811187] [.Logger] INFO/MeshConfigurationManager.swift log(message:ofCategory:withLevel:) 168 nRF lowerTransport: Access Message (akf: 0, szmic: 0, data: 0x0F8757545F85D36D85E2647000569193974EA701) received
2023-10-31 20:00:16.141482+0530 IosBasicConfigurationTest[1837:811187] [.Logger] INFO/MeshConfigurationManager.swift log(message:ofCategory:withLevel:) 168 nRF upperTransport: Upper Transport PDU (encrypted data: 0x0F8757545F85D36D85E2647000569193, transMic: 0x974EA701) received
2023-10-31 20:00:16.141584+0530 IosBasicConfigurationTest[1837:811266] [.Logger] WARN/MeshConfigurationManager.swift log(message:ofCategory:withLevel:) 168 nRF lowerTransport: Discarding packet (seqAuth: 7, expected > 9)
2023-10-31 20:00:16.141949+0530 IosBasicConfigurationTest[1837:811187] [.Logger] INFO/MeshConfigurationManager.swift log(message:ofCategory:withLevel:) 168 nRF access: Response Access PDU (opcode: 0x8019, parameters: 0x00060001C000000A000059000A00) received (decrypted using key: Device's Device Key)
2023-10-31 20:00:16.141538+0530 IosBasicConfigurationTest[1837:811292] [.Logger] DEBUG/MeshConfigurationManager.swift log(message:ofCategory:withLevel:) 168 nRF lowerTransport: Sending ACK (seqZero: 8, ackedSegments: 0x00000003)
2023-10-31 20:00:16.145350+0530 IosBasicConfigurationTest[1837:811292] [.Logger] INFO/MeshConfigurationManager.swift log(message:ofCategory:withLevel:) 168 nRF network: Sending Network PDU (ivi: 0, nid: 0x39, ctl: 1, ttl: 5, seq: 287, src: 0007, dst: 0006, transportPdu: 0x484A6FA8BFF588, netMic: 0x14A825737D01AFC3) encrypted using Network Key 1 (index: 0)
2023-10-31 20:00:16.146189+0530 IosBasicConfigurationTest[1837:811292] [.Logger] DEBUG/CustomBearer.swift send(_:ofType:) 46 CustomBearer TX: 003902A8DB17675C5042484A6FA8BFF58814A825737D01AFC3
```

anees17861, 2023-11-01:
We believe the issue is arising at the following line. The task.detached may cause things to be processed out of order. If it makes sense, maybe it can be moved to a serial dispatch queue instead. 
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/244781633222a9c64225d57a23a56b8a10d222c4/nRFMeshProvision/MeshNetworkManager.swift#L262-L264

xaviergspintly, 2023-11-03:
@philips77 did you find anything on this? Is there something else we can provide to you for debugging this?

philips77, 2023-11-12:
Hi,
I'm releasing version 4.1 without this change. Could you give it a spin? I have fixed some synchronization issues on the lower transport layer, which might help. If not, I'll release 4.1.1 with proposed changes.

closed 2023-11-19 by philips77