nordicsemi/IOS-nRF-Mesh-Library #589: Proxy change within a network causes the proxy filter not to be set

kkellermann, 2023-11-08:
### Where do you suspect the issue?

Issue in mesh library

### Version

4.0.1 (Latest)

### Describe the issue

The proxy filters are currently not set initially when a proxy is changed. In version 3.2.0, the function `updateProxyFilter(usingNetworkKey networkKey: NetworkKey)` in NetworkLayer.swift was called with an additional parameter for the IV-index (false by default) and thus caused `newProxyDidConnect()` to be called in ProxyFilter.swift. As this parameter is now omitted, `newProxyDidConnect()` is no longer called and the proxy filters are no longer set. However, it is described in the corresponding comments of the functions that the functions should be executed in the event of a proxy change within a network. I therefore wanted to ask here whether this is intended or whether it is a bug.

philips77, 2023-11-08:
It's already fixed in #585, which on contrary to its name also fixes some Proxy Filter issues. I'm finishing the PR and will release 4.0.2 asap.

philips77, 2023-11-08:
Basically, when the proxy is switched to another one, one needs to call `proxyDidDisconnect()` from the app.
If the proxy is disconnected, but not switched to another one, it's enough to send any message, which will trigger `.bearerClosed` error, which will clear proxy and notify the Proxy Filter.

philips77, 2023-11-16:
Fixed in 4.1.0.

closed 2023-11-16 by philips77