nordicsemi/IOS-nRF-Mesh-Library #592: I updated from version 3.1.5 to version 4.0.1. I need to update many interfaces and the changes are huge. Is there any convenient way?

liuhk8, 2023-11-13:
### Version

4.0.1 (Latest)

### Ask the question

I updated from version 3.1.5 to version 4.0.1. I need to update many interfaces and the changes are huge. Is there any convenient way?

philips77, 2023-11-13:
The biggest breaking change is minimum iOS version set to 13 due to use of `Task`, `actor`, async sequences, async methods.

philips77, 2023-11-21:
> I will try to help you here. What do you need help with?

How is the migration going?

ahmadshoh-orc, 2024-05-15:
Hi @philips77,

I recently updated from version 3.2.0 to 4.2.0 and am encountering some issues with the provisioning process. Could you please assist me with the migration?

I am developing a Flutter plugin to interact with nRFMeshProvision from Flutter. The first problem I encountered is with the provisioning process. Below is a relevant excerpt from my QBleMesh class, where I have a provision method:

```swift
import Foundation
import NordicMesh
import CoreBluetooth

public class QBleMesh {
    var connection: NetworkConnection?
    var unprovisionedDevices: [DiscoveredPeripheral] = [DiscoveredPeripheral]()
    var discoveredDevices: [DiscoveredDevice] = [DiscoveredDevice]()
    var statusEntities: [StatusEntity] = [StatusEntity]()
    
    // callbacks
    var provisionCallback: (_ provisionedNode: ProvisionedNode) -> Void = {_ in}
    var statusCallback: (_ status: StatusEntity) -> Void = {_ in}
    var discoveredDevicesCallback: (_ discoveredDevices: [DiscoveredDevice]) -> Void = {_ in}
    
    var provisioningService: ProvisioningService?
    var _meshNetworkManager: MeshNetworkManager?
    var meshNetworkManager: MeshNetworkManager {
        get {
            if self._meshNetworkManager == nil {
                _meshNetworkManager = initializeMeshNetworkManager()
            }
            return self._meshNetworkManager!
        }
    }
    
    private func initializeMeshNetworkManager() -> MeshNetworkManager {
        let meshNetworkManager = MeshNetworkManager()
        meshNetworkManager.networkParameters = .advanced { parameters in
            parameters.sarAcknowledgmentRetransmissionsCount = 2
            parameters.acknowledgmentMessageInterval = 4.2
            parameters.acknowledgmentMessageTimeout = 40.0
        }
        meshNetworkManager.logger = self
        return meshNetworkManager
    }
    
    init() {
        var meshLoaded = false
        do {
            meshLoaded = try meshNetworkManager.load()
        } catch {
            print(error)
        }
        
        if (meshLoaded) {
            meshNetworkDidChange()
        } else {
            createNewMeshNetwork()
        }
    }

    func startScan() -> Bool {
        if let networkConnection = connection {
            return networkConnection.startScanForPeripherals()
        }
        return false
    }

    func discoveredDevices(_ callback: @escaping (_ discoveredDevices: [DiscoveredDevice]) -> Void) {
        discoveredDevicesCallback = callback
        if let networkConnection = connection {
            networkConnection.discoveredPeripherals() { (discoveredPeripheral: DiscoveredPeripheral) in
                if !self.unprovisionedDevices.contains(where: { $0.peripheral.identifier == discoveredPeripheral.peripheral.identifier }) {
                    self.unprovisionedDevices.append((discoveredPeripheral.device, discoveredPeripheral.peripheral, discoveredPeripheral.rssi))
                    let device = DiscoveredDevice(address: discoveredPeripheral.peripheral.identifier.uuidString,
                                                  name: discoveredPeripheral.peripheral.name ?? "",
                                                  rssi: discoveredPeripheral.rssi)
                    if let updateIndex = self.discoveredDevices.firstIndex(where: { $0.address == device.address }) {
                        self.discoveredDevices[updateIndex] = device
                    } else {
                        self.discoveredDevices.append(device)
                    }
                }
                self.discoveredDevicesCallback(self.discoveredDevices)
            }
        }
    }
    
    func stopScan() -> Bool {
        if let networkConnection = connection {
            return networkConnection.stopScan()
        }
        discoveredDevicesCallback = {_ in}
        return false
    }
    
    func identify(_ address: String, _ callback: @escaping (_ unprovisionedNode: UnprovisionedNode) -> Void) {
        provisioningService?.identifyingComplete = callback
        if let networkConnection = connection, let uuid = UUID(uuidString: address) {
            let peripheral = networkConnection.centralManager.retrievePeripherals(withIdentifiers: [uuid]).first
            let unprovisionedDevice = unprovisionedDevices.first(where: { $0.peripheral.identifier == peripheral?.identifier })
            let deviceToIdentify = unprovisionedDevice?.device
            deviceToIdentify?.name = unprovisionedDevice?.peripheral.name
            provisioningService?.identify(deviceToIdentify, peripheral: peripheral)
        }
    }

    func provision(_ node: UnprovisionedNode, _ callback: @escaping (_ provisionedNode: ProvisionedNode) -> Void) {
        provisionCallback = callback
        self.provisioningService?.provisioningComplete = { unicastAddress, unprovisionedDevice in
            self.startConfiguration(unicastAddress, deviceToIdentify: unprovisionedDevice)
        }
        self.provisioningService?.provision()
    }
}
```

In this setup, I set the `provisioningComplete` callback for the `ProvisioningService`, which starts the configuration. The full code for `ProvisioningService` and `ConfigurationService` is provided below.

After debugging, I found that the code execution halts at the following line in the `ConfigurationService`:
```swift
_ = try? manager.send(ConfigCompositionDataGet(), to: node)
```
No response is received after this point.

Full code for `ProvisioningService` and `ConfigurationService` can be found in another comments below/

Could you please help me identify what might be causing this issue and how I can resolve it?

Thank you!

ahmadshoh-orc, 2024-05-15:
```
import Foundation
import NordicMesh

class ConfigurationService {
  
  let manager: MeshNetworkManager
  var compositionDataGetComplete: () -> Void = {}
  var configurationComplete: (_ provisionedNode: ProvisionedNode) -> Void = {_ in}
  
  init(manager: MeshNetworkManager) {
    self.manager = manager
    manager.delegate = self
  }
  
  func configure(unicastAddress: Address, name: String, address: String) {
    var provisionedNode: ProvisionedNode = ProvisionedNode(name: name, address: address)
    
    if let meshNetwork = self.manager.meshNetwork {
      if let node = meshNetwork.node(withAddress: unicastAddress) {
        if let appKey = meshNetwork.applicationKeys.first {
          _ = try? manager.send(ConfigAppKeyAdd(applicationKey: appKey), to: node)
          if node.isCompositionDataReceived {
            setupModels(&provisionedNode, unicastAddress: unicastAddress)
          } else {
            _ = try? manager.send(ConfigCompositionDataGet(), to: node)
            compositionDataGetComplete = {
              self.setupModels(&provisionedNode, unicastAddress: unicastAddress)
            }
          }
        }
      }
    }
  }
  
  func setupModels(_ provisionedNode: inout ProvisionedNode, unicastAddress: Address) {
    if let meshNetwork = self.manager.meshNetwork {
      if let node = meshNetwork.node(withAddress: unicastAddress) {
        if let appKey = meshNetwork.applicationKeys.first {
 
          if let primaryElement = node.primaryElement {
            provisionedNode.primaryElement = configurePrimaryElement(primaryElement, appKey: appKey)
          }
          
          for element: Element in node.elements {
            provisionedNode.segments.append(configureSegment(element, appKey: appKey))
          }
        }
      }
    }
    
    self.configurationComplete(provisionedNode)
  }
  
  /// configure the primary element of the node
  func configurePrimaryElement(_ element: Element, appKey: ApplicationKey) -> PrimaryElement {
    let primaryElement: PrimaryElement = PrimaryElement(unicastAddress: element.unicastAddress)

    if let batteryModel = element.model(withSigModelId: .genericBatteryServer) {
      if let batteryConfig = ConfigModelAppBind(applicationKey: appKey, to: batteryModel) {
        _ = try? self.manager.send(batteryConfig, to: element.parentNode!)
      }

      let localAddress: MeshAddress = MeshAddress(self.manager.localElements.first!.unicastAddress)
      let publishMessage = Publish(to: localAddress, using: appKey, usingFriendshipMaterial: false, ttl: 8, period: Publish.Period(TimeInterval(15)), retransmit: Publish.Retransmit())
      if let publicationSet = ConfigModelPublicationSet(publishMessage, to: batteryModel) {
        _ = try? self.manager.send(publicationSet, to: element.parentNode!)
      }
    }

[... 221 lines cut ...]
                    )
                    self.identifyingComplete(unprovisionedNode)
                } else {
                    self.identifyOrOpenBearer();
                }
            }
        case .complete:
            // provisioning is done so we can close the connection to the gatt.
            do {
                try self.bearer?.close()
            } catch {
                print("bearer.close error after update to 4.2.0")
            }
            break
        case .failed(_):
            // provisioning failed so we should inform the user about the failure.
            self.abortProvisioning()
            break
        default:
            break
        }
    }
}

```

philips77, 2024-05-15:
Hello,
When the provisioning is completed, you call `self.bearer?.close()`. This will close the GATT connection with the device. At that point the device is switching to normal mode. There are 2 ways you can deal with it:
1. Reconnect, discover services and proceed with configuration
2. Handle https://developer.apple.com/documentation/corebluetooth/cbperipheraldelegate/peripheral(_:didmodifyservices:) method and rediscover services.

Anyway, you need to be connected to any provisioned Node before starting to send messages. I can't find reconnection in your code. 

In case of nRF Mesh app, there's `NetworkConnection` class, which constantly scans for the device if it is not connected and connectes to first found matching device (when Auto Connect is enabled).

In the latest version of nRF Mesh app it is also possible to reconnect immediately to the device (even if one was connected) to be able to configure Low Power Nodes (LPN), which don't have any Friend yet. By directly connecting to them using GATT the client can send what's needed without relying on Friends. 
Have a look at:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/267216832aaa19ba6ffa1b49720a34fd3c2f8072/Example/Source/View%20Controllers/Network/Provisioning/ProvisioningViewController.swift#L392-L401

ahmadshoh-orc, 2024-05-15:
Thank you for your answer.

I have an extension for the QBleMesh class that facilitates the creation of a new MeshNetworkManager instance. The NetworkConnection class, derived from the nRFMesh app example with some small additional features, should ensure control over the GATT connection, ensuring it's opened or closed appropriately. Or am i missing something?

```swift
extension QBleMesh {
  func createNewMeshNetwork() {
    let provisioner = Provisioner.init(name: UIDevice.current.name)
    _ = meshNetworkManager.createNewMeshNetwork(withName: "bt-control", by: provisioner)
    _ = meshNetworkManager.save()

    meshNetworkDidChange()
  }
  
  func meshNetworkDidChange() {
    let meshNetwork = meshNetworkManager.meshNetwork!
    let primary: Element = Element(name: "PrimaryElement", location: .first, models: [
      Model(sigModelId: .genericOnOffClient, delegate: GenericOnOffClientDelegate(meshNetworkManager, statusManager: self)),
      Model(sigModelId: .lightCTLClient, delegate: LightCTLClientDelegate(meshNetworkManager, statusManager: self)),
      Model(sigModelId: .genericBatteryClient, delegate: GenericBatteryClientDelegate(meshNetwork, statusManager: self)),
//      Model(vendorModelId: .healthModelId, companyId: .companyId, delegate: VendorHealthClientDelegate(meshNetworkManager))
      Model(vendorModelId: .timerModelId, companyId: .companyId, delegate: VendorCountdownClientDelegate(meshNetworkManager))
    ])
    
    meshNetworkManager.localElements = [primary]
    
    setApplicationKey()
    setupModelApplicationKeysForLocalNode(meshNetwork.applicationKeys.first!)

    print("meshNetwork.nodes \(meshNetwork.nodes)")
    print("meshNetwork.groups \(meshNetwork.groups)")
    
    connection = NetworkConnection(to: meshNetwork)
    connection?.isConnectionModeAutomatic = true
    connection?.dataDelegate = meshNetworkManager
    connection?.logger = self
    meshNetworkManager.transmitter = connection
    connection?.open()
    
    provisioningService = ProvisioningService(meshNetworkManager: meshNetworkManager)
    meshNetworkManager.delegate = self
  }
  
  func setApplicationKey() {
    let meshNetwork = meshNetworkManager.meshNetwork!
    if(meshNetwork.applicationKeys.isEmpty) {
      let defaults = UserDefaults.standard
      
      var applicationKey:Data? = nil
      
      if let storedKey = defaults.object(forKey: "ApplicationKey") as? Data {
        applicationKey = storedKey
      }
      
      if applicationKey == nil {
        applicationKey = Data.random128BitKey()
        defaults.set(applicationKey!, forKey: "ApplicationKey")
      }
      
      do {
        _ = try meshNetwork.add(applicationKey: applicationKey!, name: "\(UIDevice.current.name)")
      } catch {
        log(message: "Application Key could not be added", ofCategory: .access, withLevel: .error)
      }
    }
  }
  
  func setupModelApplicationKeysForLocalNode(_ applicationKey: ApplicationKey) {
    // first setup battery client
    if let localElements = meshNetworkManager.meshNetwork?.localProvisioner?.node?.elements {
[... 321 lines cut ...]
      centralManager.scanForPeripherals(withServices: [MeshProxyService.uuid], options: nil)
    }
    if proxies.isEmpty {
      isOpen = false
      delegate?.bearer(self, didClose: nil)
    }
  }
  
  func bearerDidConnect(_ bearer: Bearer) {
    if !isOpen, let delegate = delegate as? GattBearerDelegate {
      delegate.bearerDidConnect(bearer)
    }
  }
  
  func bearerDidDiscoverServices(_ bearer: Bearer) {
    if !isOpen, let delegate = delegate as? GattBearerDelegate {
      delegate.bearerDidDiscoverServices(bearer)
    }
  }
  
  func bearer(_ bearer: Bearer, didDeliverData data: Data, ofType type: PduType) {
    dataDelegate?.bearer(self, didDeliverData: data, ofType: type)
  }
}
```