nordicsemi/IOS-nRF-Mesh-Library #595: How to migrate a custom model from library version 3.2.0 to 4.1.0?

FluffyBunniesTasteTheBest, 2023-11-17:
### Version

4.1.0 (latest)

### Ask the question

Do custom models require any modifications when updating the library from version 3.2.0 to 4.1.0? If so, which? 

**Some more details:**
The app is in production since about one and a half years and it is working well with library version 3.2.0.  Now, after migrating to version 4.1.0 our custom model behaves a little funny:

1. Replies from the custom model are now received through `MeshNetworkDelegate.didReceiveMessage(),` rather than `ModelDelegate.didReceiveResponse()` Well, a little copy&past could easily fix it, question is, is this intended? 
2. When some data comes in, the following message is logged: ```Local Vendor Model model on Primary Element (0x0001) not bound to key: Application Key 1 (index: 0)``` Interesting is, that the reply/status message is properly received through `MeshNetworkDelegate.didReceiveMessage()`. Beside that, `ConfigModelAppBind` is sent to the local model during app initialization.

Again, everything was working well with the version 3.2.0, so it seems like I've overlooked some crucial steps. So any ideas on what to consider when migrating a custom model to library version 4.1.0 are very much appreciated. 

Thank you very much for your help - and for providing this library!

philips77, 2023-11-18:
> Do custom models require any modifications when updating the library from version 3.2.0 to 4.1.0? If so, which?

No, nothing should change here. Perhaps only thing that vendor messages should extend either `UnacknowledgedVendorMessage`, `AcknowledgedVendorMessage` or `VendorResponse`, but perhaps you made that change already.

> 1. Replies from the custom model are now received through `MeshNetworkDelegate.didReceiveMessage()`, rather than `ModelDelegate.didReceiveResponse()` Well, a little copy&past could easily fix it, question is, is this intended?

`MeshNetworkDelegate.didReceiveMessage()` reports all received messages, while `ModelDelegate.didReceiveResponse()` reports messages targeting the given model, i.e.:
- Model must define the op code on it's [`messageTypes`](https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/293210d4827ab865462f70aa32daa9493cb13fbd/nRFMeshProvision/ModelDelegate.swift#L68) list
- Model must be bound to the App Key which the message is encrypted with.

> 2. When some data comes in, the following message is logged: Local Vendor Model model on Primary Element (0x0001) not bound to key: Application Key 1 (index: 0) Interesting is, that the reply/status message is properly received through MeshNetworkDelegate.didReceiveMessage(). 

That means, that the second condition isn't met. The local vendor model seems not to be bound to the App Key.

> Beside that, ConfigModelAppBind is sent to the local model during app initialization.

Now, that is the real issue. This seems not to work. Check if you're binding the right App Key to the correct Model, check if after that call the model is bound (you may export network configuration and check JSON).

philips77, 2023-11-18:
Btw, that works in the nRF Mesh app. After setting up local models I can control the state of the LED remotely.

closed 2023-11-21 by philips77