nordicsemi/IOS-nRF-Mesh-Library #599: Send message await forever

jack8525, 2024-01-04:
### Where do you suspect the issue?

Issue in mesh library, e.g. packets being dropped

### Version

4.1.0 (latest)

### Describe the issue

`_ = try? await globalMeshNetworkManager?.send(meshMessage, to: MeshAddress(sendMessageTask.address), using: applicationKey)`
I send a AcknowledgedMeshMessage
but now bearDidClose
[image]
this else no handle
so await forever
how can I do

philips77, 2024-01-04:
Thank you for reporting the issue. I'll look into it and come back to you.

philips77, 2024-01-04:
For Acknowledged messages there's a timer on Access Layer that should try to retransmit the message and time out eventually.

philips77, 2024-01-04:
OK. I see where the issue may occur. 

AccessLayer sets the repeat & timeout timer only for acknowledge messages sent to a Unicast Address:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/293210d4827ab865462f70aa32daa9493cb13fbd/nRFMeshProvision/Layers/Access%20Layer/AccessLayer.swift#L239-L244

That means, that acknowledged messages sent to a Group Address (or Virtual Address) don't get that timer set. Such messages may be sent to 0+ nodes and the transmitter can receive 0+ responses. In that case, it is the underlying layer that should notify the network manager, just like for unacknowledged messages.

However, as you pointed out, when the Acknowledged Mesh Message is unsegmented, like `ConfigOnOffSet`, and sending the message fails, the network manager is never notified:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/293210d4827ab865462f70aa32daa9493cb13fbd/nRFMeshProvision/Layers/Lower%20Transport%20Layer/LowerTransportLayer.swift#L257-L260

I'll prepare a fix.

philips77, 2024-01-04:
[image]

philips77, 2024-01-04:
Btw, for multicast addresses it is recommended not to use Acknowledged messages, but their Unacknowledged counterparts. This significantly decreases the traffic in the network.

philips77, 2024-01-04:
Hmm... I did some more digging and seems like the Network Layer doesn't throw `bearerDidClose` when the message is sent to a group address, as such message may be handled internally:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/293210d4827ab865462f70aa32daa9493cb13fbd/nRFMeshProvision/Layers/Network%20Layer/NetworkLayer.swift#L156-L167

This is the `shouldLoopback` method:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/293210d4827ab865462f70aa32daa9493cb13fbd/nRFMeshProvision/Layers/Network%20Layer/NetworkLayer.swift#L444-L450

As you're getting that error, that means, that you're sending the message to a Unicast Address. In that case, the network manager should wait until the response is received from the target, which happens on Access Layer.
I'll leave #600 open, but don't think it's required.

philips77, 2024-01-04:
Could you say what message are you trying to send exactly, what are your network parameters (default?) and what is the destination address of the message?

jack8525, 2024-01-05:
Thank you for your help.

philips77, 2024-01-05:
There will be a timeout. The timeout is set using:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/293210d4827ab865462f70aa32daa9493cb13fbd/nRFMeshProvision/Layers/NetworkParameters.swift#L219-L230
[Documentatuion](https://nordicsemiconductor.github.io/IOS-nRF-Mesh-Library/documentation/nrfmeshprovision/networkparameters/config/discardacknowledgedmessages(after:))

jack8525, 2024-01-06:
`func send(_ message: AcknowledgedMeshMessage,
              from localElement: Element? = nil, to model: Model,
              withTtl initialTtl: UInt8? = nil) async throws -> MeshResponse`
I use that func when send AcknowledgedMeshMessage.

jack8525, 2024-01-06:
I found failedToSendMessage is not handle.Beacase Address change to MeshMessage.

jack8525, 2024-01-06:
[image]
Can I handle deliveryCallbacks here?