nordicsemi/IOS-nRF-Mesh-Library #610: Receiving UnknownMessage with OpCode 0x008260 When Publishing Current State of LightCTL

Ahmadshoh, 2024-03-06:
### Version

3.2.0

### Ask the question

When attempting to publish the current state of LightCTL, the iOS app receives an UnknownMessage with opcode 0x8260 instead of the expected LightCTLStatus message opcode (0x8260).

Below is logs from the Chip.

This log snippet indicates successful publication of BLE Mesh messages, including one with opcode 0x8260. However, the iOS app receives an UnknownMessage instead of the expected LightCTLStatus message.

Expected Outcome:
The iOS app should correctly interpret the published message as a LightCTLStatus message with opcode 0x8260 when attempting to publish the current state of LightCTL.

### Relevant log output

```text
(905658) LAMP_STATE_RESP: Received lamp state for lamp 1 with on: 1, level: 25, temp: 255 
I (905678) LAMP_STATE_RESP: Generic on off error code: ESP_OK 
I (905678) LAMP_STATE_RESP: Ctl publish error code: ESP_OK 
I (905688) BLE_Mesh: Published successfully
I (905688) BLE_Mesh: 0x3fc953c0   82 04 01                                          |...|
I (905698) BLE_Mesh: Published successfully
I (905698) BLE_Mesh: 0x3fc9640b   82 60 ff 3f 9c 63 00 00                           |.`.?.c..|
I (905708) DEMO-MODE: New light level: 16383
I (910028) BAT_RESP: Received battery response with 1 
I (910028) DEMO-MODE: Sent battery update with soc: 0, type: 2, state 0

Logs from App:

Access PDU (opcode: 0x8260, parameters: 0xFF3F9C630000) received (decrypted using key: Devs iPhone6+ (index: 0))
received a message -> UnknownMessage(opCode: 0x008260, parameters: FF3F9C630000)
```

philips77, 2024-03-06:
You will find more info here: https://nordicsemiconductor.github.io/IOS-nRF-Mesh-Library/documentation/nordicmesh/localnode

Ahmadshoh, 2024-03-06:
Yes. I did all these steps. Here are some codes from my project.

extension QBleMesh {
  func createNewMeshNetwork() {
    let provisioner = Provisioner.init(name: UIDevice.current.name)
    _ = meshNetworkManager.createNewMeshNetwork(withName: "bt-control", by: provisioner)
    _ = meshNetworkManager.save()
    
    meshNetworkDidChange()
  }
  
  func meshNetworkDidChange() {
    let meshNetwork = meshNetworkManager.meshNetwork!
    let primary: Element = Element(name: "PrimaryElement", location: .first, models: [
      Model(sigModelId: .genericOnOffClient, delegate: GenericOnOffClientDelegate(meshNetworkManager, statusManager: self)),
      Model(sigModelId: .lightCTLClient, delegate: LightCTLClientDelegate(meshNetworkManager, statusManager: self)),
      Model(sigModelId: .genericBatteryClient, delegate: GenericBatteryClientDelegate(meshNetwork, statusManager: self)),
//      Model(vendorModelId: .healthModelId, companyId: .companyId, delegate: VendorHealthClientDelegate(meshNetworkManager))
      Model(vendorModelId: .timerModelId, companyId: .companyId, delegate: VendorCountdownClientDelegate(meshNetworkManager))
    ])
    
    meshNetworkManager.localElements = [primary]
    
    setApplicationKey()
    setupModelApplicationKeysForLocalNode(meshNetwork.applicationKeys.first!)

    print("meshNetwork.nodes \(meshNetwork.nodes)")
    print("meshNetwork.groups \(meshNetwork.groups)")
    
    connection = NetworkConnection(to: meshNetwork)
    connection?.isConnectionModeAutomatic = true
    connection?.dataDelegate = meshNetworkManager
    connection?.logger = self
    meshNetworkManager.transmitter = connection
    connection?.open()
    
    provisioningService = ProvisioningService(meshNetworkManager: meshNetworkManager)
    meshNetworkManager.delegate = self
  }
  
  
  And here is LightCTLClientDelegate class
  
  /*
 * Copyright (c) 2019, Nordic Semiconductor
 * All rights reserved.
 *
 * Redistribution and use in source and binary forms, with or without modification,
 * are permitted provided that the following conditions are met:
 *
 * 1. Redistributions of source code must retain the above copyright notice, this
 *    list of conditions and the following disclaimer.
 *
 * 2. Redistributions in binary form must reproduce the above copyright notice, this
 *    list of conditions and the following disclaimer in the documentation and/or
 *    other materials provided with the distribution.
 *
 * 3. Neither the name of the copyright holder nor the names of its contributors may
 *    be used to endorse or promote products derived from this software without
 *    specific prior written permission.
 *
 * THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS" AND
 * ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED
 * WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE DISCLAIMED.
 * IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT,
 * INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT
 * NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR
 * PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY,
 * WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE)
 * ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
 * POSSIBILITY OF SUCH DAMAGE.
 */

import Foundation
import nRFMeshProvision

class LightCTLClientDelegate: ModelDelegate {
  
  let meshNetworkManager: MeshNetworkManager
  let statusManager: StatusUpdate
  let messageTypes: [UInt32 : MeshMessage.Type]
  let isSubscriptionSupported: Bool = true
  
  var publicationMessageComposer: MessageComposer? {
    func compose() -> MeshMessage {
      return LightCTLSetUnacknowledged(lightness: UInt16(((Double(self.state) / 100.0) * 65535).rounded()), temperature: 10400, deltaUV: 0)
    }
    let request = compose()
    return {
      return request
    }
  }
  
  /// The current state of the Light CTL Client model.
  var state = 25 {
    didSet {
      publish(using: self.meshNetworkManager)
    }
  }
  
  init(_ meshNetworkManager: MeshNetworkManager, statusManager: StatusUpdate) {
    let types: [GenericMessage.Type] = [
      LightCTLStatus.self
    ]
    messageTypes = types.toMap()
    self.meshNetworkManager = meshNetworkManager
    self.statusManager = statusManager
  }
  
  // MARK: - Message handlers
  
  func model(_ model: Model, didReceiveAcknowledgedMessage request: AcknowledgedMeshMessage,
             from source: Address, sentTo destination: MeshAddress) -> MeshMessage {
    fatalError("Not possible")
  }
  
  func model(_ model: Model, didReceiveUnacknowledgedMessage message: MeshMessage,from source: Address, sentTo destination: MeshAddress) {
    if let status = message as? LightCTLStatus {
      let luminosity = UInt16(((Double(status.lightness) / 65535) * 100).rounded())
      statusManager.update(unicastAddress: source, onOff: nil, luminosity: luminosity, temperature: status.temperature, powerStatus: nil, onlineStatus: true)
    }
  }

  func model(_ model: Model, didReceiveResponse response: MeshMessage,
             toAcknowledgedMessage request: AcknowledgedMeshMessage,
             from source: Address) {
    if let status = response as? LightCTLStatus {
      let luminosity = UInt16(((Double(status.lightness) / 65535) * 100).rounded())
      statusManager.update(unicastAddress: source, onOff: nil, luminosity: luminosity, temperature: status.temperature, powerStatus: nil, onlineStatus: true)
    }
  }
}

philips77, 2024-03-06:
In that case I suspect either invalid data, or a bug in the data parser here:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/6369af8dc42d0ae0f48ea5c4e893b9d3b8e10686/Library/Mesh%20Messages/Lighting/LightCTLStatus.swift#L112-L127

Your data is `FF-3F-9C-63-00-00`, which is 6 bytes. This will fail on the first line. When `nil` is returned, the library defaults to `UnknownMessage`.

Let's compare with the spec...

philips77, 2024-03-06:
[image]

Note, that OpCode is not included here, so the data in fact should be 4 or 9 bytes long.

Ahmadshoh, 2024-03-06:
Ok. Thank you very much. I will try to solve the problem using this data. I'll let you know about the results.