nordicsemi/IOS-nRF-Mesh-Library #611: nRFMeshProvision/iOSDFULibrary: "Multiple commands produce PrivacyInfo.bundle" when creating an archive.

FluffyBunniesTasteTheBest, 2024-04-15:
### Version

4.2.0 (latest)

### Ask the question

After updating nRFMeshProvision (4.2.0) and iOSDFULibrary (4.15.1), creating an archive now fails with:

```
Multiple commands produce '/Users/x/Library/Developer/Xcode/DerivedData/Runner-detzyouqxapgtmabgpoqdafgagau/Build/Intermediates.noindex/ArchiveIntermediates/Runner/IntermediateBuildFilesPath/UninstalledProducts/iphoneos/PrivacyInfo.bundle'
Target 'iOSDFULibrary-PrivacyInfo' (project 'Pods') has create directory command with output '/Users/x/Library/Developer/Xcode/DerivedData/Runner-detzyouqxapgtmabgpoqdafgagau/Build/Intermediates.noindex/ArchiveIntermediates/Runner/IntermediateBuildFilesPath/UninstalledProducts/iphoneos/PrivacyInfo.bundle'
Target 'nRFMeshProvision-PrivacyInfo' (project 'Pods') has create directory command with output '/Users/x/Library/Developer/Xcode/DerivedData/Runner-detzyouqxapgtmabgpoqdafgagau/Build/Intermediates.noindex/ArchiveIntermediates/Runner/IntermediateBuildFilesPath/UninstalledProducts/iphoneos/PrivacyInfo.bundle'
```

Do you have any ideas how to fix or workaround this?
Thank you.

philips77, 2024-04-15:
Oh.. no, I have no idea. I was hoping that Xcode should handle that. Now it is recommended for all libraries to provide their _PrivacyInfo_ files, so they should be merged with the app's _PrivacyInfo_ file.

Perhaps I could do similar as here:
https://github.com/getsentry/sentry-cocoa/pull/3651

philips77, 2024-04-15:
Could you try updating Nordic DFU library to 4.15.2 and check if that's resolved?

FluffyBunniesTasteTheBest, 2024-05-02:
The issue disappeared after updating iOSDFULibrarz from 3.15.1 to 4.15.3. 

Thank you very much for your help, and especially for providing those libraries!

closed 2024-05-02 by FluffyBunniesTasteTheBest