nordicsemi/IOS-nRF-Mesh-Library #612: Need help with configuring nodes

ahmadshoh-orc, 2024-05-21:
### Version

4.2.0 (latest)

### Ask the question

Hi everyone,

After updating the library version from 3.2.0 to 4.2.0, I am encountering some issues. I am developing a Flutter plugin that utilizes the NordicMesh library. After provisioning a node, I aim to directly configure its model for use with my app. However, I am not receiving any response when trying to get ConfigCompositionData. Could anyone help me identify the problem? According to the logs, I have a bearer connection open in the NetworkConnection. Below are some relevant snippets of code from my plugin. The NetworkConnection class is an exact copy from the nRFMesh example app.

I know this is a long post with many lines of code, but I'm having trouble understanding the issue and how to solve it. I'm new to this library and IoT, so any help would be greatly appreciated.

### Base Class of Plugin
This class handles the initial configurations and creates instances of `MeshNetworkManager` and `NetworkConnection`.

```swift
public class QBleMesh {
    var connection: NetworkConnection!
    
    var _meshNetworkManager : MeshNetworkManager!
    var meshNetworkManager: MeshNetworkManager {
        get {
            if self._meshNetworkManager == nil {
                _meshNetworkManager = initializeMeshNetworkManager()
            }
            
            return self._meshNetworkManager!
        }
    }

    var unprovisionedDevices: [DiscoveredPeripheral] = [DiscoveredPeripheral]()
    var discoveredDevices: [DiscoveredDevice] = [DiscoveredDevice]()
    var scanCallback: (() -> Void) = {}

    var centralManager: CBCentralManager!
    var provisioningService: ProvisioningService?

    private func initializeMeshNetworkManager() -> MeshNetworkManager {
        let meshNetworkManager = MeshNetworkManager()
        
        meshNetworkManager.networkParameters = .basic { parameters in
            parameters.setDefaultTtl(5)
            parameters.discardIncompleteSegmentedMessages(after: 10.0)
            parameters.transmitSegmentAcknowledgmentMessage(
                usingSegmentReceptionInterval: 0.06,
                multipliedByMinimumDelayIncrement: 2.5)
            parameters.retransmitSegmentAcknowledgmentMessages(
                exactly: 1, timesWhenNumberOfSegmentsIsGreaterThan: 3)
            parameters.transmitSegments(withInterval: 0.06)
            parameters.retransmitUnacknowledgedSegmentsToUnicastAddress(
                atMost: 2, timesAndWithoutProgress: 2,
                timesWithRetransmissionInterval: 0.200, andIncrement: 2.5)
            parameters.retransmitAllSegmentsToGroupAddress(exactly: 3, timesWithInterval: 0.250)
            parameters.retransmitAcknowledgedMessage(after: 4.2)
            parameters.discardAcknowledgedMessages(after: 40.0)
        }
        
        meshNetworkManager.logger = self
        
        return meshNetworkManager
    }
    
    override init() {
        super.init()
        centralManager = CBCentralManager()

        var meshLoaded = false
        do {
            meshLoaded = try meshNetworkManager.load()
        } catch {
[... 462 lines cut ...]
Access Message (akf: 0, szmic: 0, data: 0x331C36AA53FAEF636302) received (decrypted using key: Primary Network Key (index: 0))
Upper Transport PDU (encrypted data: 0x331C36AA53FA, transMic: 0xEF636302) received
Response Access PDU (opcode: 0x8003, parameters: 0x00000000) received (decrypted using key: DEMO_2_AC's Device Key)
ConfigAppKeyStatus(networkKeyIndex: 0, applicationKeyIndex: 0, status: Success) received from: 0002
message comes here ConfigAppKeyStatus(networkKeyIndex: 0, applicationKeyIndex: 0, status: Success)

### Chip Logs

I (14723948) BLE_Mesh: ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT, bearer PB-GATT
W (14724008) BLE_MESH: No Health Server context provided
W (14724338) BLE_MESH: No Health Server context provided
I (14724338) DevKey: 41 7f f0 7b 08 86 8e d0 df e9 87 43 19 07 ff 06 
I (14724498) BLE_Mesh: ESP_BLE_MESH_NODE_PROV_LINK_CLOSE_EVT, bearer PB-GATT
I (14724508) NetKey: fa ba 3d 20 ba 96 36 6b 33 82 28 e9 ad 04 c5 22 
I (14724508) NID: 5f 
I (14724508) EncKey: bd 5c a8 69 da ef 1e df be be 2c c1 3c 24 b1 c1 
I (14724518) NetID: ba 8f fe 06 cd d9 63 1f 
I (14724518) BLE_Mesh: ESP_BLE_MESH_NODE_PROV_COMPLETE_EVT
I (14724528) BLE_Mesh: net_idx: 0x0000, addr: 0x0002
I (14724528) BLE_Mesh: flags: 0x00, iv_index: 0x00000000
W (14724578) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x13
I (14726498) BLE_Mesh: ESP_BLE_MESH_MODEL_OP_APP_KEY_ADD
I (14726498) BLE_Mesh: net_idx 0x0000, app_idx 0x0000
I (14726498) AppKey: 59 c3 4c f8 52 0d 80 20 61 44 28 7b c0 ed 27 6b
```

philips77, 2024-05-22:
It's only possible to send one message to a device at a time. You'd have to wait for the App Key Status message before sending Composition Data Get. 

Have a look here: 
```swift
            if let node = meshNetwork.node(withAddress: unicastAddress) {
                if let appKey = meshNetwork.applicationKeys.first {
                    _ = try? manager.send(ConfigAppKeyAdd(applicationKey: appKey), to: node)

                    if node.isCompositionDataReceived {
                        setupModels(&provisionedNode, unicastAddress: unicastAddress)
                    } else {
                        compositionDataGetComplete = {
                            self.setupModels(&provisionedNode, unicastAddress: unicastAddress)
                        }
                        _ = try? manager.send(ConfigCompositionDataGet(), to: node)
                    }
                }
            }
```
If you support iOS 13+ try calling these method with `await`, this should to the trick: 
```swift
            if let node = meshNetwork.node(withAddress: unicastAddress) {
                if let appKey = meshNetwork.applicationKeys.first {
                    _ = await try? manager.send(ConfigAppKeyAdd(applicationKey: appKey), to: node)

                    if node.isCompositionDataReceived {
                        setupModels(&provisionedNode, unicastAddress: unicastAddress)
                    } else {
                        compositionDataGetComplete = {
                            self.setupModels(&provisionedNode, unicastAddress: unicastAddress)
                        }
                        _ = await try? manager.send(ConfigCompositionDataGet(), to: node)
                    }
                }
            }
```
I don't remember the Swift syntax, but something like that. You need to create a `Task` to run it, add they are synchronous.

philips77, 2024-05-22:
Or `try? await`.

ahmadshoh-orc, 2024-05-23:
Thank you for your suggestion. Everything is working fine now.

closed 2024-05-23 by ahmadshoh-orc