nordicsemi/IOS-nRF-Mesh-Library #614: How to set scene for 2nd  element?

DuanPhan92, 2024-06-14:
### Version

4.2.0 (latest)

### Ask the question

Scenes always set scene id for 1st element. I can’t set scene for 2nd element.

philips77, 2024-07-24:
Hello,
Sorry for the late response.

What do you mean exactly? Are you sending Scene Store message from the phone to your device, or try to store a scene in the app?

Note, that it's possible for a Node to have Scene Server and Scene Setup Server models on more than one element. Each such pair controls states on that element and all elements until the element with the next pair.

DuanPhan92, 2024-07-25:
Here's the translated response:

---

Hello,

My device has 4 elements, and each element has both the Scene Server and Scene Setup Server models installed. However, the app currently only allows setting scenes for the first element, and the other elements cannot have scenes set. Could you please help check this issue?

Thank you!

philips77, 2024-07-25:
OK, now I understand. 

Indeed, instead of having controls in Scene Server and Scene Setup Server model views, like for other models, the app allows to control scenes from a designated screen. The intention was to make the UI more intuitive, as some actions require the first and some the other model. But it also assumes that the server models are located on the main element of a node and does not currently allow to use other scene registers.

I think the easiest fix would add a "Scenes" row on the Node screen for each element with Scene Server. The address of that element would be passed to the existing view, so it doesn't have to be changed.

Let's keep this issue open as feature request. I hope to fix it this year. I'm busy with other projects at the moment so I can't promis any date. If you want to contribute, please create a PR and I'll be more than happy to review and merge!

oluwaoh, 2025-03-17:
bluetooth get on and off without notifications