nordicsemi/IOS-nRF-Mesh-Library #615: Why is the Bluetooth device json table data in my configuration table sometimes inexplicably cleared? I hope to reply to me

liuhk8, 2024-06-17:
### Version

4.2.0 (latest)

### Ask the question

Why is the Bluetooth device json table data in my configuration table sometimes inexplicably cleared? I hope to reply to me

philips77, 2024-06-17:
Hello,
The mesh configuration doesn't clear on its own. You may be creating a new mesh network and saving it at some point, or not calling `manager.save()` after it has been set up.
The network can also be remotely cleared by another Node if it knows the Device Key of the phone's local node.

philips77, 2024-06-17:
I highly doubt remove node reset is what you observe, by twe way.

liuhk8, 2024-06-18:
I didn't call the node removal interface to handle it. It was often very sudden, and the device list was cleared. I suspected that some actions would clear the node collection in the json table. I found that one possible cause of this problem is to create two mesh networks in quick succession, which is likely to clear the node collection of the first one.

philips77, 2024-07-24:
Yes, that perhaps may cause such behavior, if you use the same `Storage` impl for both. They can be overwritten and cleared in the process.