nordicsemi/IOS-nRF-Mesh-Library #616: Health messages not handled

JulesDommartin, 2024-07-09:
### Version

4.2.0 (latest)

### Ask the question

Hi, I was trying to get health messages, but It seems like the library won't let me override the nil delegate that is set in the init of the localElements.

[image]
This function will automatically remove my ModelDelegate for Health Server.

This is very important for our project to get this Health messages, could you please provide a solution for that problem?

Thanks in advance, and thanks for all of your work!

### Relevant log output

```text
GattBearerLog: <- 0x001C5C87818292BA321CBDE26178D44302467F0461073127
2024-07-09 15:23:37.511293+0200 Runner[831:81366] [Network] Network PDU (ivi: 0, nid: 0x1C, ctl: 0, ttl: 8, seq: 38, src: 0033, dst: C000, transportPdu: 0xBDE26178D44302467F04, netMic: 0x61073127) received
2024-07-09 15:23:37.516336+0200 Runner[831:81366] [LowerTransport] Access Message (akf: 1, aid: 0x24, szmic: 0, data: 0x2E51D8FF05233BFBE0) received (decrypted using key: Primary Network Key (index: 0))
2024-07-09 15:23:37.522755+0200 Runner[831:81366] [UpperTransport] Upper Transport PDU (encrypted data: 0x2E51D8FF05, transMic: 0x233BFBE0) received
2024-07-09 15:23:37.526008+0200 Runner[831:81366] [Access] Access PDU (opcode: 0x 4, parameters: 0x00AA0402) received (decrypted using key: Application Key 1 (index: 0))
Default message received
UnknownMessage(opCode: 0x000004, parameters: 00AA0402)
UnknownMessage
```

philips77, 2024-07-12:
Hello,
Indeed, the library does what you say. The plan was to implement all models that require Device Key encryption inside the library, but due to lack of resources we had to switch to other projects. Health models are still in the backlog, but now DFU will be the priority. 

As a solution I would recommend importing the library locally, via SPM or CocoaPods, and adding Health client and server delegates like they are for Config and Scenes. We could gladly accept a PR :)

It's vacation period now, so the fastest we could do anything about it would be September-ish, so the above solution is best for you, if technically possible.

JulesDommartin, 2024-10-28:
Hi @philips77, how are you going ?
Do you have any update on this ? If not, are you planning on doing it ? 
It is hard to find the time for us to work on this but if you don't have the time too, we can try.

Thanks in advance.

philips77, 2024-10-30:
Hello,
I'm starting to work on DFU soon, then I'm planning Subnet Bridge and Health messages later. It's day mid 2025 or later. I'm quite busy with other projects as well... :(

ALEKSANDER FILIP NOWAKOWSKI | R&D Engineering Manager - Mobile Applications
P +48 668 007 947 | Warsaw, Poland
nordicsemi.com<http://www.nordicsemi.com/> | devzone.nordicsemi.com<https://devzone.nordicsemi.com/>

JulesDommartin, 2024-10-31:
Thanks for your answer, I will probably start implementing it myself and submit a PR in a few times. Good luck with your planning !

JulesDommartin, 2024-11-05:
#626 Here is the PR for this !