nordicsemi/IOS-nRF-Mesh-Library #622: Retransmission

mpiffari, 2024-10-09:
### Version

4.2.0 (latest)

### Ask the question

Hi Dev Team,

I'm reaching out to ask about the approach used to handle BLE write failures during mesh operations (like provisioning, configuration, etc.). Specifically, is there a way to configure the number of retransmissions for a single packet in case of errors?

philips77, 2024-10-09:
Have a look at this file:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/main/Library/Layers/NetworkParameters.swift
It allows to control all network parameters defined in Bluetooth Mesh Protocol 1.1.

philips77, 2024-10-09:
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/28ff22e8ad636c664e91ceceaef8796ce1a33633/Library/MeshNetworkManager.swift#L73 Use this to customioze.

closed 2024-10-09 by philips77