nordicsemi/IOS-nRF-Mesh-Library #623: Provisioner assignment of node/address to self

eanderlind, 2024-10-11:
### Version

4.2.0 (latest)

### Ask the question

Is there a technical reason as to why Provisioner is restricting the src address it is assigning itself to be within it's range and locking uuid at creation?
It appears it would be sufficient to check that it is using an address from any node that it has assigned itself, rather than creating one per Provisioner instance.
Of course if it needs to create a new node for this, the most convenient is to assign from within it's assigned unicast range.

There is an associated low probability security and usability issue with this when Provisioner is exported. Let me know if interested in obtaining more details offline.

philips77, 2024-10-13:
Hello,
The provisioner ranges are specified in [Mesh Configuration Database Profile](https://www.bluetooth.com/specifications/specs/mesh-configuration-database-profile-1-0-1/), however the restriction you mentioned is something we added ourselves.

> It appears it would be sufficient to check that it is using an address from any node that it has assigned itself, rather than creating one per Provisioner instance.

This is not possible due to Replay Protection Count. Each node must keep track of the last sequence number received from any Unicast Address it receives a message to prevent replay attack. A phone cannot send messages "in the name of another node", therefore it need to have its own address = be a Node.

> Of course if it needs to create a new node for this, the most convenient is to assign from within it's assigned unicast range.

We have chosen this approach. It is not specified, as I said, but this seems logical. Another option would be to use the address of a provisioner that creates the other provisioner object, but such provisioner can be initially created without an address (so it can only provision, but can't send messages, so can't configure). But if later the provisioner decides to configure - it needs an address, so it's only safe to assign an address from its own range. Hence the restriction.

> There is an associated low probability security and usability issue with this when Provisioner is exported. Let me know if interested in obtaining more details offline.

Yes, when exported, some other implementations may have a different/no restriction.

Also, as I said before, remember that a provisioner doesn't have to have an address at all. Then it can only provision nodes, that is assign them the address + initial Network Key.

eanderlind, 2024-10-13:
> This is not possible due to Replay Protection Count. Each node must keep track of the last sequence number received from any Unicast Address it receives a message to prevent replay attack. A phone [...]

Agree that in order to send mesh msgs it needs an own address. Still possible as described as long as node is explicitly allocated to the phone, ie a different phone doesn't use same Node record. Current code, from what I can tell, saves per address seq (ie only part of IVseq) as UserData. 

> We have chosen this approach. It is not specified, as I said, but this seems logical.

Perhaps logical, but as mentioned is not secure if Provisoner has an address. One approach that would be secure is for Provisoner to store in UserData which Node (with address) it has created for itself. Whenever create a new Provisoner record or add configuration capabilities, check whether UserData shows the Phone has a Node record with addresss that it itself created, or create one and update UserData with a reference to the new record.

> Yes, when exported, some other implementations may have a different/no restriction.

As mentioned, low probability but dire security implications of reusing the Provisioner's node address (see cryptography two-time PAD).

Another complication (in NetworkLayer) is that on IVupdate, the Provisioner currently only resets the IVseq for the currently active localProvisoner, not for any other Provisoner records that it may have created and used with separate addresses. 
In addition it seems, as mentioned above, UserData+SeqAuth doesn't store (at least part of) the IV-part in the per-address UserData (as specified in spec) and thus will not know when IV has changed for a particular source host. (If true this may be a separate issue - should have been uncovered if perform a test IVupdate, so perhaps code elsewhere that addresses this.)

eanderlind, 2024-11-04:
Thanks for the above clarifications, especially the dual use of UserData+SeqAuth for both sending and receiving PDUs - this wasn't clear to us at first. We got all the info needed and were able to redesign library to mitigate security issue when more than one device/phone may act as Provisioner by keeping track of whether Provisoner node was created locally or remotely. It was also straightforward to address the IVupdate case when more than one Provisioner has been defined for the phone. From our perspective issue can be closed.

>     As mentioned, low probability but dire security implications of reusing the Provisioner's node address (see cryptography two-time PAD).
> 
> That's again, not a security issue. Nodes will ignore messages sent with "not high enough" sequence number. If there are 2+ nodes in the network with the same unicast address, only the messages sent by the one with higher sequence number will be processed by other nodes.

Because Mesh security is based on a stream cipher, the above packets that reuse a src SEQ# pair are sufficient for an attacker to decrypt the pair of PDUs, and (with some additional social engineering and active attack techniques) also forge packets using same keys (incl device key). It appears you assume everyone involved operates on an honors system to never use a Provisioner on more than one device, in which case are all good. This is reasonable assumption for some networks. Attacks are simpler to execute in larger multi-hop network configurations and where CDB data sharing events can be predicted or detected.

closed 2024-11-04 by eanderlind

philips77, 2024-11-05:
It is possible to export CDB without Device Keys, especially Device Keys of other Provisioners.
The rule should be that there's one Provisioner (with optionally one Node) per phone/device and it can only be used there.

If someone gets keys, either by getting CDB, or reading from any device - the security is lost anyway.

We leave to the 3rd party apps which use our libraries for manage provisioners and sharing CDB. Each case is different.
I would assume, that when a new user is created, it gets assigned a new Provisioner object with a new Unicast Address. If it's a super user, which can provision new devices - the ranges are non-empty. If that's a regular user that should only be allowed controlling states - the ranges are empty (impl detail: Unicast Address range should contain local addresses) and CDB doesn't contain Device Keys. 
Apps should not allow users to modify ranges, reprovision, etc, if not required, and should not allow reading CDB as json or obtaining it any other way, so it should be sent encrypted. But that's all an implementation detail. nRF Mesh app allows all of that, as well as the library.