nordicsemi/IOS-nRF-Mesh-Library #625: Abnormal timing of hardware data acquisition

zjl-col, 2024-10-24:
### Version

4.2.0 (latest)

### Ask the question

43/10000
实时翻译
划译
Hello, while using the latest version for timed communication with hardware, occasional crashes may occur. Could you please help guide me? Thank you!
[image]
[image]

philips77, 2024-10-24:
Thanks for creating the issue. I'll fix it soon.

zjl-col, 2024-11-01:
Hello, can you reproduce this situation?

philips77, 2024-11-03:
I didn't yet yet, but what you provided is enough.

zjl-col, 2025-01-22:
Hello, the above issue is that sending data is too long, and the probability of unicast transmission occurring for a long time is higher. Adjusting the amount of data sent now can avoid this issue. There is a new question that needs to be consulted. During normal communication, occasional disconnections may occur. If the communication time is too long, there may be situations where the connection cannot be made. For example, during normal communication, if a callback is received but the next task has not been triggered, output a log:
[image]

After a period of time, the NetworkConnection will be triggered func bearer(_ bearer: Bearer, didClose error: Error?) {} 

May I ask where the problem occurred during my configuration process?thank you