nordicsemi/IOS-nRF-Mesh-Library #627: Sometimes MeshNetworkManager.proxyFilter.proxy does not get set when loading a previously configured network

tylerjames, 2024-11-20:
### Where do you suspect the issue?

Issue in mesh library, e.g. packets being dropped

### Version

4.2.0 (latest)

### Describe the issue

I'm not sure exactly how to reproduce this but I found it when I needed to send a message to _any_ connected node to be relayed to other nodes in the network.

I wanted to find a node that was connected and so I used `MeshNetworkManager.proxyFilter.proxy` and it worked fine. Except eventually it stopped working, and then sometimes would start working again.

I found that it was because _sometimes_ but not always `MeshNetworkManager.proxyFilter.proxy` would be `nil` even though `NetworkConnection.proxies` contained several open bearers. I'm not sure why this is the case. It looks like sometimes the `NetworkLayer.handle(proxyConfigurationPdu:)` will sometimes fail to find a node here:
```swift
let proxyNode = meshNetwork.node(withAddress: proxyPdu.source)
networkManager.proxy?.handle(message, sentFrom: proxyNode)
```
And it will send `nil` and the the `ProxyFilter.proxy` gets set to `nil`

I'm thinking that maybe me using `ProxyFilter.proxy` in this way is not a reliable way to get a connected node. If that's the case what is a better way? I know I can see open bearers in `NetworkConnection.proxies` but I'm not sure how to associate them with the corresponding `Node` so that I can use one of the `MeshNetworkManager.send()` variants to send a message to the Node.

Any help would be appreciated.

philips77, 2024-11-21:
Hello,
Do you have all the possible proxy Nodes in your database? The spec says that it may be null if such address isn't found in the network.
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/ef2db414a2a2037f5e815ea5703838e571474ecf/Library/ProxyFilter.swift#L217-L220
Could you provide logs from _Console_ app (filter with `subsystem=com.nordicsemi.nRFProvision`) or Xcode logs when it happens? I wonder what's going on there.

philips77, 2025-06-07:
Hi,
Since version 4.3 the `ProxyFiltrer.proxy` gets set to `UnknownNode` object if there is no Node witht its Unicast Address in the Mesh Configuration Database.

Could you tell if this has improved for you?