nordicsemi/IOS-nRF-Mesh-Library #630: Some time device not supported error happening

sapna908, 2025-01-22:
### Version

3.2.0

### Ask the question

we have integrated lib to our project, and some time  getting "device not supported" error 
It is coming randomly and also on latest ios 18 connections are dropping 
Can some one help resolve this issue 

### Relevant log output

```text

```

philips77, 2025-01-22:
Hello,
This may be a caching issue. If you're flashing the DKs with some other firmware in between mesh the phone may cache services of the other fw and not clear it. Sometimes also the service discovery returns 0 services, but I saw it only on Android.
It's hard to say what's the reason for your issue. The error is thrown when the service discovery returns no Mesh services.

oluwaoh, 2025-03-17:
sometimes it get debugg and on without my notice