nordicsemi/IOS-nRF-Mesh-Library #633: New release

JulesDommartin, 2025-01-27:
### Version

4.2.0 (latest)

### Ask the question

Hello, when can we imagine to have a new version of the SDK ? There are several features that block us for commercial release (especially the Health feature) that are waiting since july of previous year. 

Thank you in advance !

### Relevant log output

```text

```

philips77, 2025-01-27:
Hello!
I resumed work on this project. Next thing is the DFU, I already have the Firmware Update and Firmware Distribution messages in #58. 
I think I can add Health messages as well, this doesn't seem to be a big thing (?).

philips77, 2025-01-27:
Oh, right. I'll try to quickly add more Health messages and will push the release soon.

JulesDommartin, 2025-01-27:
Thanks for your answer ! Have a great day 😃

philips77, 2025-01-29:
Hello,
Please, follow #634.

closed 2025-03-25 by philips77