nordicsemi/IOS-nRF-Mesh-Library #654: After version 4.2.0 cannot send messaged from iPhone thats not the main provisioner

lecniers, 2025-02-25:
### Where do you suspect the issue?

Mesh related issue in nRF Mesh, e.g. sending vendor messages doesn't work

### Version

4.2.0 (latest)

### Describe the issue

I'm importing the mesh configuration from a JSON file, and on version 4.2.0, sending messages after connecting to a proxy node is working, but on version 4.3.0 or 4.3.1, it is broken.

After debugging the Example App  I see that after connecting to a proxy node, the proxyFilter.proxy is always nil, my mesh is already provisioned I'm just importing a JSON mesh config on a guest phone.
 
Because this change to the MeshNetworkManager send function:
```
       // A message may be sent to a local Node, or using a GATT Proxy Node.
        // Check if the message can be relayed to the destination using a Proxy Node.
        // The Proxy Node must know the Network Key; otherwise it will not be able to
        // decode the destination and decrement TTL.
        if destination.address.isUnicast {
            guard localNode.contains(elementWithAddress: destination.address) ||
                  proxyFilter.proxy?.knows(networkKey: applicationKey.boundNetworkKey) == true else {
                print("Error: The GATT Proxy Node is not connected or it cannot decrypt \(applicationKey.boundNetworkKey)")
                throw AccessError.cannotRelay
            }
        } 
```

Inside ProxyFilter:

```
   /// The connected Proxy Node.
    ///
    /// This is `nil` if no GATT Proxy Node is connected, or the connected Node is
    /// not in the local mesh configuration database.
    public private(set) var proxy: Node?
```

### Relevant log output

```text

```

philips77, 2025-03-03:
The question is: why is `proxyFilter.proxy` set to `nil` when you are connected? 
Do you have the Proxy Node in your network configuration?

lecniers, 2025-03-03:
@philips77 The proxy node is not in the JSON network configuration, the phone is not the provisioner, every node is provisioned using a Python program over USB, the phone is just a guest on the mesh to check health status for other nodes, topology, it doesn't know about any nodes on the mesh just the network key, app key and mesh name. I'm just using the JSON config cuz is the only way to bind network keys to app keys for models.

philips77, 2025-03-04:
OK, I get it.
Thanks for the feedback. I'll restore the previous functionality in the library (meaning, if `proxy` is `nil`, it will still try to send the message).
I may check the proxy in the app only, as my main goal was to give more relaible user experience for the most common case when message times out just because the proxy is not connected.

PR #656 (2025-03-25): Bugfix: Allowing sending message through an unknown proxy node
This PR fixes #654.

It adds a new type `UnknownNode`, which is set as `ProxyFilter.proxy` when an unknown Node is connected as GATT proxy. Upon receiving a `ProxyConfigurationMessage` (`FilterStatus`) from a connected Proxy the phone will know its Unicast Address and the Network Key. These data are then used to initialize the `UnknownNode`.

When a message is sent through such Node the library will always use this Network Key, as it's unaware[^1] of other keys that the Proxy Node may know. 

[^1]: In theory, it gets a Secure Netwo [... cut]
closed 2025-03-25 by philips77